github0null/eide #233: Request to Contact the Creator

vmwsree, 2023-04-10:
Hi there,

I hope you are doing well. I am writing to inquire about the possibility of hiring or onboarding you as a consultant in a project we are working on. I believe this could be a mutually beneficial opportunity for both of us with monitory benifits. If you are interested, please contact me at vmwsree@gmail.com.

I look forward to hearing from you soon.

Best regards,

Vivek (vmwsree)

github0null, 2023-04-11:
Sorry, I have to work and don't have much time, so I can't be your consultant. If you have any questions, you can make a list and submit the issue. I will reply one by one when I have time

PS:
Not sure what you are going to do, this is a very simple plugin, it has some documentation, you can figure out how it works by simply using it, if you want to know more details, you can debug this plugin, it is open source

github0null, 2023-04-16:
expired

closed 2023-04-16 by github0null