github0null/eide #252: [bug] Warning: parse 'launch.json' error!

eatradish, 2023-05-17:
使用 EIDE 打开已有项目时，出现一个警告：

[image]

这是 项目文件夹内 `.vscode/launch.json` 的内容：

```
// https://code.visualstudio.com/docs/cpp/launch-json-reference
{
    "version": "0.2.0",
    "configurations": [
		{

			"name": "OpenOCD gdb",

			"type": "cppdbg",
			"request": "launch",

			"program": "${workspaceFolder}/build/uAptiv_Baremetal.elf",
			"args": [],
			"cwd": "${workspaceFolder}",
			"environment": [],

			"MIMode": "gdb",
			"miDebuggerPath": "mips-mti-elf-gdb",
			"miDebuggerArgs": "",
			"miDebuggerServerAddress": "localhost:3333",

			// "debugServerPath": "openocd",
			// "debugServerArgs": "-f D:/CIPUnited/OpenOCD/openocd-0.12.0-mips-2023-04-13/scripts/interface/ftdi/dp_busblaster.cfg -c \"set CPUTAPID 0x37221053\" -f D:/CIPUnited/OpenOCD/openocd-0.12.0-mips-2023-04-13/pic32mz-fixed.cfg -c \"adapter speed 2000; init; reset halt\"",
			// "serverStarted": "Listening on port 3333 for gdb connections",

			"logging": {
				"engineLogging":"verbose",
				"moduleLoad": true,
				"exceptions": true,
				"programOutput": true,
				"natvisDiagnostics": "verbose"
			},

			"targetArchitecture": "mips",
			"useExtendedRemote": true,
			"stopAtEntry": true,
			"stopAtConnect": true,

			"setupCommands": [
				{ "text": "set endian little" },
				{ "text": "set mem inaccessible-by-default off" },
				{ "text": "set disassemble-next-line on" },
				{
					"text": "-environment-cd ${workspaceFolder}",
					"description": "https://github.com/microsoft/vscode-cpptools/issues/548"
				},
				{ "text": "dir src/asm" },
			],
			"postRemoteConnectCommands": [
				{ "text": "set $pc=0xa0070000" },
			]

		}
	]
}
```

github0null, 2023-05-18:
由于用的 js 内置解析器，不支持带有注释的 json，后面会修复的

github0null, 2023-05-21:
刚刚看了一下，现在的版本已经是用的 jsonc 了

你的 json 存在问题

两个数组内的最后一项有多余的 `,` 号：

```
  { "text": "dir src/asm" }, 

  "postRemoteConnectCommands": [
				{ "text": "set $pc=0xa0070000" },
			]
```

github0null, 2023-05-22:
因为插件目前用的库 jsonc 只支持标准的 json

jsonc-parser 有一定的容错处理

[image]

这个库适合 读和解析 内容，但没有 `stringify` 这种 api

Crystal-RainSlide, 2023-06-12:
@github0null 

`jsonc.stringify()` 基本就是 `JSON.stringify()`：
https://github.com/onury/jsonc/blob/adc4a7a801e73bc13a89e911e05d8f409ac5d4c8/lib/jsonc.js#L219

- 为保证与 VSCode 的兼容性，最好用 VSCode 用的、微软方维护的 `jsonc-parser`；
- 为保留格式、注释等内容，最好用 `jsonc.write()`，`jsoncParser.applyEdits()` 等基于修改的 API，来得到要保存回文件的字符串。
- 还有就是，保存前，可能最好再读取一下目标文本文件，和原先读取的对比一下，一样再继续，确保不会弄丢期间发生的意外修改。

github0null, 2023-06-12:
applyEdits 适合用来配合 vscode 的编辑器 api 来修改指定位置的内容，

applyEdits 用来合并有改动的部分，

```js
export function applyEdits(text: string, edits: EditResult): string;

export type EditResult = Edit[];

export interface Edit {
    /**
     * The start offset of the modification.
     */
    offset: number;
    /**
     * The length of the modification. Must not be negative. Empty length represents an *insert*.
     */
    length: number;
    /**
     * The new content. Empty content represents a *remove*.
     */
    content: string;
}
```

插件需要修改 json 对象内容，如果要在带有注释的基础上，再生成正确的 EditResult，太复杂

这样一通操作下来，目的仅仅是为了保留 json 中的注释

---

这个 DebugConfigurationGenerator 在早先用于生成默认的调试配置模板，

对于新手来说，在一些调试配置模板上只用改几个字段即可开始调试，要不要注释显得无关紧要

对于已经熟悉 debug 的用户来说，这个功能可能多此一举，且会干扰到自己手写的配置，

后面可能加入设置项来做一个开关，可以选择 启用 或 禁用 该功能，这样就不会影响到自己纯手写的配置

Crystal-RainSlide, 2023-06-13:
`jsonc-parser` 同时提供的 `modify()` 方法即可生成 `applyEdits()` 所需的 `EditResult`。不过，看下来，修改成使用这些方法确实是比较麻烦的。

那么，可以考虑忽略掉注释，把 `jsonc` 分别换成 `jsonc-parser` 的 `parse()` 方法和 `JSON.stringify()`，这样一来，既不会因为尾随逗号等导致读取错误，也不用大改。

github0null, 2023-06-14:
好的，既然它会生成这些参数，后面会逐渐替换到 jsonc-parser 的

github0null, 2024-06-10:
v3.17.2 中，调试配置生成功能已替换为使用 jsonc-parser

closed 2024-06-10 by github0null