github0null/eide #253: [question] 增加 make 支持

Crystal-RainSlide, 2023-05-19:
我们想为 [make](https://www.gnu.org/software/make/) 增加 make 支持，目前，大部分嵌入式软件项目都可以转换成 EIDE 项目，用 EIDE 的构建工具构建，但这对于用户而言增加了额外的工作量，同时，也有少数例外情况。考虑到绝大多数 C/C++ 项目使用 make，如果增加 make 支持，就能解决这些问题。

至于实现的话，我们会先简单加一个“使用 make 构建”右键菜单；但是，如果好做的话，就在合适的地方加一个选项，允许项目切换构建工具，在 EIDE 构建工具和 make 之间选择。

Crystal-RainSlide, 2023-05-19:
（另外，在 v3.9.1 中被移除的生成 Makefile 功能，在现有的两个 nls JSON 中残余了两个字段，应当已经可以删除：
``` JS
"eide.project.gen.makefile"
"settings.builder.gen.makefile.params"
```）

github0null, 2023-05-20:
这个功能可以有，但完美实现比较困难

我之所以不做，我觉得实际做出来，也可能并没有人用，因为用 make 的，基本自己就直接命令行编译了，或者用了 eclipse，而且对于大多数开发单片机的用户来说，其实很多不会用 make, 也不懂, 也不会去选择 make

> 您对此功能有何看法？

有几个疑问：
 - 我不知道你们打算如何实现 makefile 的生成，确保每种工具链，都能生成正确的 rules，并且能够兼容用户写的 makefile
 - 软件需要跨平台，不知道你们打算支持几种 make:  mingw-make / gnu make / nmake ... ?
 - 对于 make 环境本身，按哪种方式提供，是需要用户提前预装，还是插件提供二进制版本
 - 可能需要实现 makefile 解析器，因为用户可能会添加 makefile，需要从中获取出编译参数，传递给 c/c++ 插件做 intellisense

> 您愿意接受增加此功能的 PR 吗？

如果能完美实现，且不影响现有功能，当然可以

> 如果愿意的话，您希望此功能如何实现？

我并没有什么想法，我觉得一个开发环境只应该使用一种 构建系统，这样能够降低软件的复杂度，减少问题，毕竟构建工具本身的作用都一样，都是为了生成 目标文件

一般如果别人需要用 make, 我会推荐它用 `makefile tools` 插件，这样更纯粹和简单

closed 2023-05-27 by github0null

Crystal-RainSlide, 2023-06-12:
感谢您的回复！

我们这边之所以想加上一个简单的 Make 支持，一方面是鉴于大量现有的项目已经在使用 make 来构建，转移到 EIDE 构建有成本和可能的困难；另一方面，也正是考虑到其他 Make 支持拓展要么较为复杂、自成体统（Makefile Tools），或者没有解决好你提出的问题（所有其他 Make 支持插件），我们不认为它们是多么通用、多么十全十美的解决方案。

Makefile 确实是十分棘手，无论是生成，还是后续更改。正因如此，我们也不打算涉及这些，而是只求能够便捷地运行 Make。

如果一个项目已有 Makefile，就不需要生成了，后续修改也是用户自己的事情。但有，这样就无需用户配置 EIDE 构建，或者为运行一个 Make 而费心编写 `.vscode/tasks.json`，再或者把使用习惯（从简单地执行 Make）切到 Makefile Tools。

而如果一个项目没有 Makefile，与其在 EIDE 中重新支持生成 Makefile，可以优先让人使用更好配置一些的 EIDE 构建。这时，如果用户乐意，或者项目必须考虑没有 EIDE 的开发环境等等，也可自行编写 Makefile。

至于 Make 的种类、环境（尤其是 Shell），提供方式等等，我看到现有的这些解决方案也没有多涉及这些，都是让用户自行指定 Make 二进制路径，没指定就 fallback 到 PATH 中的 `make`。