github0null/eide #264: [question] Model JSON 中如何标注仅接受自然整数的参数？

MingcongBai, 2023-06-14:
**What you want to do**

正在扩充 MIPS MTI GCC 类型项目的编译器选项。

**Describe the problem**

目前 [在 mips.mti.gcc.model.json 中登记了一些额外的 MIPS MTI GCC 选项](https://github.com/mips-studio/eide/blob/cip/update-mips-mti-gcc-model/res/data/models/unix/mips.mti.gcc.model.json#L130) ，但其中 `-G` 选项后面是要跟随自然整数（0 和正整数）的，目前似乎没有这样的选项类型？目前的实现是作为一个 `string`，但显然很容易造成问题。

如有此类参数类型，请指正。

**Expected behavior**

可标注 `-G` 参数后跟随自然整数（0 和正整数）。

github0null, 2023-06-15:
unify_builder 处理 model.json 的键值都是按字符串处理的，bool 和 number 也会直接转成 string

因为后续生成的命令也是字串型的，这样处理起来更简单一些

[image]

---

如果要限制用户的输入，其实这个 model 直接设置成 "value" 型就行了

因为 UI 是根据 `mips.mti.gcc.verify.json` 里的定义生成的，可以在 `mips.mti.gcc.verify.json` 改字段的类型

[image]

只不过，目前的 UI 没有适配 `number` 类型，需要添加类似于这样的分支才能支持：

https://github.com/github0null/eide_options_view/blob/master/src/App.vue#L198-L207

MingcongBai, 2023-06-16:
了解，这边研究下如何实现

closed 2024-06-07 by github0null