github0null/eide #265: [question] Error due to ... folder ownership?

SynonZelra, 2023-06-22:
**What you want to do**
Dev on CH582

**Describe the problem**
New Project -> Remote Respo. -> WCH -> CH582 -> random name -> build

`/bin/bash: -c: line 1: unexpected EOF while looking for matching `''
/bin/bash: -c: line 2: syntax error: unexpected end of file`

**Expected behavior**
I've just noticed that if I create the project to, say /mnt which ownership is root, it will ask if I trust the content, and build works just fine.
If I create the project to some other folder which ownership is my current user, no trust asked, and return the problem output on build.

SynonZelra, 2023-06-23:
Sry I need more test

github0null, 2024-01-12:
You can not use these Special symbols: '{,},(,),&,|,...' in filename or path

because these symbols will be interpreted as part of the command, will cause a commandline error

closed 2024-01-12 by github0null