github0null/eide #272: [question] RISCV的LD文件和download配置问题

Antecer, 2023-07-20:
```
// 默认LD配置
MEMORY
{
	FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 256K
	RAM (xrw) : ORIGIN = 0x20000000, LENGTH = 64K
}
```
```
// 修改后的LD配置
MEMORY
{
	FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 64K
	CONST (rx) : ORIGIN = 0x00010000, LENGTH = 192K
	RAM (xrw) : ORIGIN = 0x20000000, LENGTH = 64K
}
SECTIONS
{
// .text 后面添加
	.const :{
	. = ALIGN(4);
		KEEP(*(.user_const_data))
	. = ALIGN(4);
	}>CONST AT >CONST
}
```

```
//download.cmd默认配置
set HEXFILE=%1
set "HEXFILE=%HEXFILE:\=/%"

openocd -f ./tools/wch-interface.cfg -f ./tools/wch-target.cfg -c init -c halt -c "flash erase_sector wch_riscv 0 last " -c "program %HEXFILE%" -c "verify_image %HEXFILE%" -c wlink_reset_resume -c exit

// 官方推荐修改LD后的配置(测试使用MounRiver下载成功)
openocd.exe  -f wch-riscv.cfg   -c page_erase -c init -c halt    -c "program    xxx.hex/elf/bin verify "  -c exit
```

问题是，在EIDE，使用修改LD后的配置下载程序会验证失败，以下是错误信息(希望能得到解决办法):
```
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-01-03-10:00)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-Link-CH549  mode:RV version 2.8 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901125
[wch_riscv.cpu.0] Target successfully examined.
Info : starting gdb server for wch_riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
** Programming Started **
Info : device id = 0x263e578f
Info : flash size = 256kbytes
Info : ROM 256 kbytes RAM 64 kbytes
Info : Flash write discontinued at 0x00006218, next section at 0x00010000
** Programming Finished **
** Verify Started **
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:1162: Error: ** Verify Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1223
at file "embedded:startup.tcl", line 1162
Error: [wch_riscv.cpu.0] Failed to flush registers. Ignoring this error.

 *  终端进程“cmd.exe /C "D:\MounRiver\MounRiver_Studio\toolchain\OpenOCD\bin\openocd.exe -s "h:\MajorProjects\AmusingDeviceProject\AntDevice\AntDevice" -f tools/wch-interface.cfg -f tools/wch-target.cfg -c "program \"h:/MajorProjects/AmusingDeviceProject/AntDevice/AntDevice/build/Debug/AntDevice.hex\" verify" -c "reset run" -c "exit""”已终止，退出代码: 1。 
```

github0null, 2023-07-20:
> cmd.exe /C "D:\MounRiver\MounRiver_Studio\toolchain\OpenOCD\bin\openocd.exe -s "h:\MajorProjects\AmusingDeviceProject\AntDevice\AntDevice" -f tools/wch-interface.cfg -f tools/wch-target.cfg -c [...]

下载命令不对，烧录器配置不要选用 OpenOCD

烧录器配置应该选用 Shell Flasher，直接用的 download.cmd 里的命令烧录的，

因为 CH32V 是定制的烧录命令和openocd，不能直接用普通的 openocd 命令去烧录

Antecer, 2023-07-20:
原来如此，问题解决了，非常感谢！

Antecer, 2023-07-20:
刚看到Shelll有个从云端安装Shell烧录脚本
[image]
于是好奇的试了一下，然后就下载失败了，错误提示：
```
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Warn : Bypassing JTAG setup events due to errors
** Programming Started **
Error: couldn't open h
embedded:startup.tcl:1136: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1201
at file "embedded:startup.tcl", line 1136
```

好奇心害死猫！
最后我还是把shell改回原来的配置，就好了。

但是我还是觉得，这个问题应该反馈一下，毕竟是从插件内直接下载的配置，竟然无法使用。

github0null, 2023-07-20:
请粘贴完整的命令行 log

Antecer, 2023-07-20:
> 

```
--------------------------------------------------
firmware: h:/MajorProjects/AmusingDeviceProject/AntDevice/AntDevice/build/Debug/AntDevice.hex
--------------------------------------------------
Open On-Chip Debugger 0.11.0+dev-02215-gcc0ecfb6d-dirty (2022-09-28-11:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Ready for Remote Connections
Info : WCH-Link-CH549  mod:RV version 2.8 
Info : wlink_init ok
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Warn : Bypassing JTAG setup events due to errors
Info : [riscv.cpu.0] datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40901125
[riscv.cpu.0] Target successfully examined.
Info : starting gdb server for riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
Info : device id = 0x263e578f
Info : flash size = 256kbytes
Info : ROM 256 kbytes RAM 64 kbytes
erased sectors 0 through 255 on flash bank 0 in 0.212998s

Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Warn : Bypassing JTAG setup events due to errors
** Programming Started **
Error: couldn't open h
embedded:startup.tcl:1136: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1201
at file "embedded:startup.tcl", line 1136

 *  终端进程“cmd.exe /C "bash ./tools/program_flash.sh "h:\MajorProjects\AmusingDeviceProject\AntDevice\AntDevice\build\Debug\AntDevice.hex""”已终止，退出代码: 1。 
```

github0null, 2023-07-20:
把命令改成 `bash ./tools/program_flash.sh "./${OutDirBase}/${ProjectName}.hex"` 试试？

可能是 shell 模式的路径解析问题

Antecer, 2023-07-20:
问题解决，能正常下载了，的确是相对路径的问题。
但是这样写的话，程序文件那个配置似乎就没用了吧。
感觉应该把这个相对路径的父级写到shell里面，这样在gui菜单配置参数能简单点。

github0null, 2023-07-20:
只是验证问题原因，后面会修改 shell 脚本的，命令还是保持不变

Antecer, 2023-07-20:
好的，大佬解决问题真快！

closed 2023-07-20 by Antecer