github0null/eide #279: [question] .NET6 runtime not found on kubuntu22.04LTS

wieniawski-lai, 2023-08-20:
**What you want to do**
When startup VSCode, it is reported `Not found .NET6 runtime on your pc. please install it`

**Describe the problem**
I have installed
- vscode
- embedded IDE plugin
- dotnet6 in `/opt/donet`, and envirioment variable is also set:
```shell
export DOTNET_ROOT=/opt/dotnet
export PATH=$PATH:/opt/dotnet
```
and I can find `dotnet` by type `dotnet --version` in shell.

**Expected behavior**
How could I startup vscode without that reporting?

CodonNine, 2023-10-28:
Check your .NET runtime version,EIDE only support for .NET6.0.You can try to remove your previous .NET runtime,then run sudo apt-get install -y dotnet-runtime-6.0.

github0null, 2024-04-07:
Use `dotnet --list-runtimes` list your runtimes. 

We just support .NET v6.x

closed 2024-04-07 by github0null