github0null/eide #283: [bug] 安装实用工具报错：Error : certificate has expired

Lingling-LynnChan, 2023-09-03:
我试图在新安装的EIDE中安装实用工具

我已经完成了工具链的配置，在选择安装实用工具后，在下拉列表中选择任意一个工具（例如Cppcheck），右下角会弹出一个警告窗口并终止下载，内容为`Error : certificate has expired`。目前尝试过重装、启动代理、关闭代理。问题没有得到解决。

[image]

**Desktop (please complete the following information):**
 - OS: Windows11
 - EIDE Version 3.13.2023061101
 - VSCode Version 1.81

没有log

Abrillant-Lee, 2023-09-03:
我也遇到相同问题, 目前也尚未解决

github0null, 2023-09-04:
证书过期了，近几天会重新申请的

----

临时解决方法：

打开设置，取消勾选，关闭掉插件的默认代理：

[image]

这个代理选项是为了访问 `raw.githubusercontent.com` 这个域名的（因为国内这个域名不挂梯子访问不了），

如果取消勾选，你就要自己挂梯子，然后再点击安装 Cppcheck

Lingling-LynnChan, 2023-09-04:
> 证书过期了，近几天会重新申请的
> 
> 临时解决方法：
> 
> 打开设置，取消勾选，关闭掉插件的默认代理：
> 
> [image]
> 
> 这个代理选项是为了访问 `raw.githubusercontent.com` 这个域名的（因为国内这个域名不挂梯子访问不了），
> 
> 如果取消勾选，你就要自己挂梯子，然后再点击安装 Cppcheck

问题解决了，感谢

Abrillant-Lee, 2023-09-05:
已解决, 谢谢

github0null, 2023-09-06:
证书已经重新申请了，这个代理设置项可以重新勾选了

closed 2023-09-06 by github0null

Abrillant-Lee, 2023-09-11:
收到,可以了