github0null/eide #284: [bug] 导出keil中xml缺失vendor和device内容

Deadline233233, 2023-09-14:
导出的keil文件缺失了vendor和device内容，导致用keil打开报错，提示找不到芯片。
[image]
[image]

github0null, 2023-09-21:
如果你没有安装 cmsis chip pack, 并选择芯片类型，插件是无法知道你用的何种芯片，所以无法导出这些信息给 keil

Deadline233233, 2023-10-21:
能否将pack路径固定，插件从pack读取文件。虽然大大有精简过的包，但部分芯片包还是比较大，像STM32F4就将近100M。一样的芯片每个工程都放一个包，就比较占空间了

github0null, 2024-04-07:
移动 pack路径，这个会涉及到很多改动，暂时改不了

这个导出功能，Device 和 Vendor 后面会优化一下，没有芯片型号时，根据 Arch 类型进行设置，比如：

```
          <Device>ARMCM3</Device>
          <Vendor>ARM</Vendor>
```

```       
          <Device>ARMCM0P</Device>
          <Vendor>ARM</Vendor>
```

github0null, 2024-05-13:
> 移动 pack路径，这个会涉及到很多改动，暂时改不了
> 
> 这个导出功能，Device 和 Vendor 后面会优化一下，没有芯片型号时，根据 Arch 类型进行设置，比如：
> 
> ```
>           <Device>ARMCM3</Device>
>           <Vendor>ARM</Vendor>
> ```
> 
> ```
>           <Device>ARMCM0P</Device>
>           <Vendor>ARM</Vendor>
> ```

已在 v3.16.1 中进行了调整

closed 2024-05-13 by github0null