github0null/eide #291: [bug] Cannot setup any utility tools, show "certificate has expired"

senhtry, 2023-10-13:
**What are you doing?**
Setup utility tools

**Describe the bug**
Cannot download any utility tools, show "certificate has expired"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'EIDE' tool 
2. Click on 'Setup Utillity Tools'
3. Choose any tool, for example, the "GNU Arm Embedded Toolchian"
4. See error

**Expected behavior**
Expect to automatic download tool

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win10 22H2
 - EIDE Version: v3.13.2023060401
 - VSCode Version: 1.83.1

zskevind, 2023-10-30:
Still not work after upgraded to pre-release.

ganyuanzhen, 2023-10-30:
Well, EIDE dose not connect to GitHub directly, Instead, it uses a proxy server. The certificate issue may be caused by this.

Simply set EIDE.Repository: Use Proxy to false.

This will let eide to connect to GitHub directly.

Or, edit the setting’s json file:
```
{
"EIDE.Repository.UseProxy": true
}
```
从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

closed 2023-12-19 by senhtry