github0null/eide #292: [bug] 内置Perl版本过低，会与LaTeX 需要的版本冲突

ganyuanzhen, 2023-10-19:
**What are you doing?**
Tell us what you were doing when the bug occurred.
当同时启用EIDE 与 LaTeX Workshop 时，在编写LaTeX文档后，使用latexmk 编译会使用EIDE的内置perl。由于EIDE内置的perl版本过低，会导致latexmk编译失败。因此需要关闭EIDE，十分麻烦。
**Describe the bug**
A clear and concise description of what the bug is.
由于EIDE内置的perl版本过低，会导致latexmk编译失败。因此需要关闭EIDE，十分麻烦。
**To Reproduce**
Steps to reproduce the behavior:
1. 同时安装EIDE 和 latex workshop
2. 编写latex文档
3. 使用latexmk方式编译文档
4. 报错
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
'''
File::Path version 2.08 required--this is only version 1.08 at D:/Users/15228/AppData/Local/Programs/MiKTeX/scripts/latexmk/latexmk.pl line 75.
BEGIN failed--compilation aborted at D:/Users/15228/AppData/Local/Programs/MiKTeX/scripts/latexmk/latexmk.pl line 75.
'''

当disable EIDE 后，不再报错。

**Desktop (please complete the following information):**
 - OS: Windows 11 10.0.25967
 - EIDE Version v3.13.2023061101
 - VSCode Version v1.82.1

**Additional context**
some logs, like: `<user_home>/.eide/cl.eide.log`

ganyuanzhen, 2023-10-19:
希望能更新一下perl版本(●ˇ∀ˇ●)

github0null, 2024-01-12:
这个 perl 是 msys1 里的程序。由于msys1好像没有更新了，因此暂时更新不了

closed 2024-02-27 by ganyuanzhen