github0null/eide #294: come to a strange problem when building

hiddenblue, 2023-10-25:
**What are you doing?**
build the project for stm32

**Describe the bug**
output a lot of "no such file or direcotry"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: debain12 linux kernel 6.1
 - EIDE Version 3.12.2023
 - VSCode Version 1.83.1

**Additional context**
some logs, like: `<user_home>/.eide/cl.eide.log`

[2023/10/25 15:50:0 GMT+0800] : launched
[Hidden] : spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
Error: spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
        at Object.spawnSync (node:internal/child_process:1110:20)
        at spawnSync (node:child_process:931:24)
        at Object.execFileSync (node:child_process:974:15)
        at Object.func [as execFileSync] (node:electron/js2c/asar_bundle:2:1869)
        at D.loadStm8DevList (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:7:17938)
        at EventEmitter.<anonymous> (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:7:10043)
        at EventEmitter.emit (node:events:513:28)
        at a.emit (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:1:5877)
        at /home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:20:71452
        at Generator.next (<anonymous>)
        at a (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:20:45772)

[2023/10/25 16:3:18 GMT+0800] : launched
[Hidden] : spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
Error: spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
        at Object.spawnSync (node:internal/child_process:1110:20)
        at spawnSync (node:child_process:931:24)
        at Object.execFileSync (node:child_process:974:15)
        at Object.func [as execFileSync] (node:electron/js2c/asar_bundle:2:1869)
        at D.loadStm8DevList (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:7:17938)
        at EventEmitter.<anonymous> (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:7:10043)
        at EventEmitter.emit (node:events:513:28)
        at a.emit (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:1:5877)
        at /home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:20:71452
        at Generator.next (<anonymous>)
        at a (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:20:45772)

[2023/10/25 16:4:13 GMT+0800] : launched
[Hidden] : spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
Error: spawnSync /home/chase/.eide/bin/utils/stvp_tools/stvp_utils ENOENT
        at Object.spawnSync (node:internal/child_process:1110:20)
        at spawnSync (node:child_process:931:24)
        at Object.execFileSync (node:child_process:974:15)
        at Object.func [as execFileSync] (node:electron/js2c/asar_bundle:2:1869)
        at D.loadStm8DevList (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:7:17938)
        at EventEmitter.<anonymous> (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:7:10043)
        at EventEmitter.emit (node:events:513:28)
        at a.emit (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:1:5877)
        at /home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:20:71452
        at Generator.next (<anonymous>)
        at a (/home/chase/.vscode/extensions/cl.eide-3.13.2023060401/dist/extension.js:20:45772)

ganyuanzhen, 2023-10-30:
If my memory works, the Embedded IDE only ships armclang, which should be AC6. But in your configuration, the compiler was set to AC5.

Besides, the armclang is not working outside the Embedded IDE. It nees some Envrinment vars to work properly. 

A better solution to this, if you do not need MDK-Arm pack, is to use the arm-none-eabi-gcc compiler, which is shipped with many Linux distros.

github0null, 2023-12-22:
在 TERMINAL 中，执行一下 /home/chaxxx/xxx/xxx/armcc 命令

检查你的 armcc 能否正常运行

closed 2024-01-12 by github0null