github0null/eide #299: [bug]  Compile error

sankred9527, 2023-11-29:
**What are you doing?**
I run "npm run compile" command , and get error 

**Describe the bug**
`
> eide@3.13.2023060401 compile
> tsc -p ./

node_modules/x2js/x2js.d.ts:56:27 - error TS2304: Cannot find name 'Document'.

56     xml2dom(xml: string): Document;
                             ~~~~~~~~

node_modules/x2js/x2js.d.ts:68:24 - error TS2304: Cannot find name 'Document'.

68     dom2js<T>(domNode: Document): T;
                          ~~~~~~~~

node_modules/x2js/x2js.d.ts:79:29 - error TS2304: Cannot find name 'Document'.

79     js2dom<T>(jsObject: T): Document;
                               ~~~~~~~~

src/CodeBuilder.ts:36:22 - error TS2307: Cannot find module '../lib/node-utility/File' or its corresponding type declarations.

36 import { File } from "../lib/node-utility/File";
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~
`

**To Reproduce**
Steps to reproduce the behavior:
1.  my linux machine is  arm-linux 64bit , ubuntu 22.04 
2.   
npm -v
10.2.3
3. node  -v
v20.10.0

**Expected behavior**
should compile successfully

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  ubuntu 22.04 , cpu is arm aarch64
 - EIDE Version :  git pull HEAD from eide 
 - VSCode Version [e.g. v1.84.2]

github0null, 2023-12-22:
please use `node 16.xx`

github0null, 2024-01-12:
I will close this issue, if you want to continue, please reopen it !

closed 2024-01-12 by github0null