github0null/eide #303: [bug] iar工程编译时搜索路径缺失，编译出错

luyuduan, 2023-12-31:
### 问题现象：
[image]

### 排查过程
一看是找不到头文件，但是IAR没有这个警告于是看了看两IDE 头文件搜索路径
虽然两者 头文件搜索路径 是一样的，路径下也都 **没有** 这个头文件 _#include "core_cm3.h"_
但是IAR没有报警告
[image]

然后在IAR里看了一下这个文件所在位置。看来IAR会自动添加这个搜索路径
[image]

再看看EIDE的搜索路径，缺少对这个路径的搜索
[image]

于是就报错了。

### 将就方法
怎么改我不知道，目前只好手动添加这个路径了

### 环境如下
IAR ANSI C/C++ Compiler V8.30.1.114/W32 for ARM

Embedded IDE v3.13.2023060401

github0null, 2024-01-12:
cmsis core header 是外部的库，需要你自己和代码放一起的，你可以按如下步骤进行安装

[image]

-------------

IAR/Keil 为了方便会帮你自动引用这个路径，但这样你的工程总是有缺失的，因为一旦离开 iar/keil，你的工程就会少这些头文件

luyuduan, 2024-01-17:
嗯，方法有很多，都可以解决

closed 2024-01-31 by luyuduan