github0null/eide #306: [bug] EIDE在编译较大项目后VSCode短暂卡死

ElluIFX, 2024-01-08:
我理解这很大概率不是EIDE的问题，但我想问问作者是否有解决方案

具体触发场景是当我编译一个较大的项目后，VSCode会卡住几秒钟，这段时间内eide插件无响应，VSCode前端界面正常响应并显示一个加载条，但intelliSense不响应。

尤其是当我使用Build and Flash时，编译后需要近十秒才会触发烧录，个人猜测是C/C++插件在读取build文件夹内的生成项目并分析，因为编译出的warning和error只有在恢复正常后才会更新到VSCode界面中

这实在影响我的使用体验，因此我想问问是否有解决方案

我尝试了下面的设置，但C/C++插件仍然尝试分析build输出

```json
{
    "C_Cpp.files.exclude": {
        "**/build": true,
        "**/build/**": true,
        "build": true,
        "**/.vscode": true,
        "**/.vs": true
    },
    "C_Cpp.exclusionPolicy": "checkFilesAndFolders",
}
```

**Desktop (please complete the following information):**
1. VSCode
版本: 1.85.1
提交: 0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2
日期: 2023-12-13T09:49:37.021Z
浏览器: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Code/1.85.1 Chrome/114.0.5735.289 Electron/25.9.7 Safari/537.36
2. EIDE
版本: v3.13.2023060401
3. C/C++
版本: v1.18.5
4. 编译器
Armclang

github0null, 2024-01-12:
@ElluIFX 

不清楚你的项目有多少文件

你可以按照该文档 https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

抓取一下从 开始卡死 到 结束卡死 的 profile 文件，然后把该文件提交上来

ElluIFX, 2024-01-12:
感谢回复，我用profile找到了问题，是gruntfuggly.todo-tree这个扩展引起的，实在是令人难以理解
我关闭这个issue了