github0null/eide #316: [question] import 导入 uvprojx文件后没有生成对应的工程，也没有任何提示，这个是什么原因呢？

chengdaqiang85, 2024-03-18:
**What you want to do**
我想通过导入已有的keil工程，生成对应的eide工程

**Describe the problem**
在EIDE 插件收到选择Import Project，然后选择对应的.uvprojx 文件，但是没有任何弹窗提示消息，看cl.eide.log文件中也没有任何错误输出

**Expected behavior**
希望可以根据keil工程正常生成EIDE工程

chengdaqiang85, 2024-03-18:
keil 版本
μVision V5.29.0.0
Copyright (C) 2019 ARM Ltd and ARM Germany GmbH. All rights reserved.
Toolchain:MDK-ARM Plus Version:5.29.0.0
Toolchain Path:	C:\Keil_v5\ARMIVARMCLANG\Bin
C Compiler:ArmClang.exe 	V6.13.1 
Assembler:	Armasm.exe 	V6.13.1
Linker/Locator:	ArmLink.exe 	V6.13.1
Library Manager:	ArmAr.exe 	V6.13.1
Hex Converter:	FromElf.exe 	V6.13.1
CPU DLL:	SARMCM3.DLL 	V5.29.0.0
Dialog DLL:	TCM.DLL 	V1.36.5.0
Target DLL:	SeggerUL2CM3.dll 	V2.99.36.0
Dialog DLL:	TCM.DLL		V1.36.5.0

github0null, 2024-03-29:
请把你的 .uvprojx 文件发上来

sanqiandamowang, 2024-04-18:
宝贝，我也遇到了，你是不是vsc右下角禁用通知了

closed 2024-04-20 by github0null