github0null/eide #319: 我想请问一下插件如何支持散列文件呢？[question]

DYAAL, 2024-03-29:
**What you want to do**
A clear and concise description of what you want to do.
我想请问一下插件如何支持散列文件呢？
**Describe the problem**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

github0null, 2024-03-29:
什么是散列文件？

DYAAL, 2024-03-29:
> 什么是散列文件？

请问一下方便留个v吗，我发给你看看。。
散列文件就是编写程序时候的link file， 比如：
[image]
在keil里面是通过下面这个地方去指定的，散列文件可以指示编译器去把链接文件按照要求去把变量的绝对地址放到文件要求的地方去。

[image]

github0null, 2024-03-29:
选一下这两个设置就行了

 - `Use Custom Linker Script File` 选 true
 - `Linker Script File Path` 填文件路径

在 Keil 里这个文件叫法就是 Scatter File

https://em-ide.com/docs/modules/builder

[image]

DYAAL, 2024-03-29:
okok非常感谢，再次感谢为这么好用的工具做了那么多工作

DYAAL, 2024-03-29:
> 选一下这两个设置就行了
>
> * `Use Custom Linker Script File` 选 true
> * `Linker Script File Path` 填文件路径
>
> 在 Keil 里这个文件叫法就是 Scatter File
>
> https://em-ide.com/docs/modules/builder [...]

好像是因为我的ide不知道为啥变成中文了。。导致我找不到。。它翻译为链接脚本了。。我一直都叫的散列文件。。再次感谢

closed 2024-03-29 by DYAAL