github0null/eide #321: 插件在项目保存方面的一点bug吗？[bug]

DYAAL, 2024-03-30:
**What are you doing?**
Tell us what you were doing when the bug occurred.
比如说我在插件上进行了一些配置，我也进行了保存，然而当我下次打开以后保存过的设置就不在了，如果用keil操作过的配置还是会保留，这个是什么问题，应该怎么解决呢？

1.我已经做好了配置，写好了芯片地址，安装了支持包等，
[image]

2.我确实点了保存，应该已经保存成功

[image]

3.我关闭了vscode，再重新导入
[image]
4.重新导入以后，之前的配置已经不见了，请问我有办法可以解决这个问题不？
[image]
5.主要是这个问题，使用和debug都正常，望解答
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10 xxx]
 - EIDE Version [e.g. v3.8.3]
 - VSCode Version [e.g. v1.68.1]

**Additional context**
some logs, like: `<user_home>/.eide/cl.eide.log`

github0null, 2024-03-30:
你对这个功能可能存在误解，导入项目 意味着重新新建一个工程，工程的内容是从 keil 项目中提取

https://em-ide.com/docs/getting-started/import_prj/

DYAAL, 2024-03-30:
> 你对这个功能可能存在误解，导入项目 意味着重新新建一个工程，工程的内容是从 keil 项目中提取
> 
> https://em-ide.com/docs/getting-started/import_prj/

哦哦，那如果我想打开上次保存的项目应该怎么办呢？
实际上有这样的场景，因为现在很多配置都是通过cubemx来进行配置的，而引脚要求通常会根据项目的进行有一些细微的调整，而通过cubemx保存后的项目通常一些生成的文件并没用直接加入进来，而需要重新导入。

github0null, 2024-04-01:
> 因为现在很多配置都是通过cubemx来进行配置的，而引脚要求通常会根据项目的进行有一些细微的调整，而通过cubemx保存后的项目通常一些生成的文件并没用直接加入进来

这个需要你手动添加新增的文件的

如果只是细微的调整，CubeMx 不会新添加一些新的源文件，不用重新导入

DYAAL, 2024-04-04:
> > 因为现在很多配置都是通过cubemx来进行配置的，而引脚要求通常会根据项目的进行有一些细微的调整，而通过cubemx保存后的项目通常一些生成的文件并没用直接加入进来
> 
> 这个需要你手动添加新增的文件的
> 
> 如果只是细微的调整，CubeMx 不会新添加一些新的源文件，不用重新导入

了解，非常感谢

closed 2024-04-05 by github0null