github0null/eide #332: 构建配置GCC 无法编辑RAM/FLASH布局(有需求)

Atul-8, 2024-05-07:
我今日尝试将armcc编辑器写的代码改成gcc写的代码. 然后发现 链接的时候会失败, 也即是说缺乏lds文件,由于我不会写lds文件, 再阅读你们的指导网页也没发现这个,  但是当我把lds文件给删掉时,程序工程倒是也可以编译了, 只是会出现个问题, 就是单片机的下载地址变了, 并且我不知道在哪修改,. 查来查去, 好像以前都是再ram/flash布局这里做修改的, 所以希望帮我一下拜托了

github0null, 2024-05-08:
lds 就是描述 ram/flash 布局的文件，这个文件是必须要提供的

看你用的什么芯片。一般官方的GCC例程会提供这个 lds 文件的，不用自己写；

官方的例程没有gcc的，那就不支持用gcc

Atul-8, 2024-05-08:
我今天直接用eide创建了一个新的stm32f103的工程。使用标准库。我使用gcc编译器 链接编译和下载都显示成功。但是我写的点灯程序却没能在板子上运行。如果切换成armcc编译器就可以。不知道您在测试的时候是否遇到这种情况。我还没找出原因。-------- 原始邮件 --------发件人： null ***@***.***>日期： 2024年5月8日周三 13:50收件人： github0null/eide ***@***.***>抄送： 划水佬 ***@***.***>, Author ***@***.***>主    题： Re: [github0null/eide] 构建配置GCC 无法编辑RAM/FLASH布局(有需求) (Issue #332)
lds 就是描述 ram/flash 布局的文件，这个文件是必须要提供的
看你用的什么芯片。一般官方的GCC例程会提供这个 lds 文件的，不用自己写；
官方的例程没有gcc的，那就不支持用gcc

github0null, 2024-05-09:
stm32f1 是提供了项目模板的，要是不熟悉 gcc 可以用模板新建项目

Atul-8, 2024-05-09:
昨天发现问题来源了。似乎是标准库的代码不兼容导致的。我基于stm32cubeMX生成make/cmake工程都可以直接gcc编译。不知道eide什么时候能兼容导入make工程和cmake工程？-------- 原始邮件 --------发件人： null ***@***.***>日期： 2024年5月9日周四 17:37收件人： github0null/eide ***@***.***>抄送： 划水佬 ***@***.***>, Author ***@***.***>主    题： Re: [github0null/eide] 构建配置GCC 无法编辑RAM/FLASH布局(有需求) (Issue #332)
stm32f1 是提供了项目模板的，要是不熟悉 gcc 可以用模板新建项目

github0null, 2024-05-10:
这是你的代码问题。跟你用 make / cmake 无关

closed 2024-05-10 by github0null