github0null/eide #344: 是否可以支持我们的芯片（基于SDCC）

lihebb, 2024-06-04:
**What you want to do**
[image]

EIDE是否可以在芯片选择的地方支持到我们公司的芯片？

Alphascale - 14(alp14)
Alphascale - 15(alp15)

如果可以的话，我应该提供哪些资料给您们？

github0null, 2024-06-04:
这个下拉列表选项现在是硬编码到插件里的，由于 sdcc 支持的 target 一直在增加，这个列表是不完整的，**下个版本**会改为去解析 `sdcc -v` 自动获取 target 列表

并同时增加一个 `None` 选项代表其他

```
> sdcc -v
SDCC : mcs51/z80/z180/r2k/r2ka/r3ka/sm83/tlcs90/ez80_z80/z80n/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk13/pdk14/pdk15/mos6502 4.2.0 #13081 (MINGW64)
published under GNU General Public License (GPL)
```

**因此只要你们定制版的 sdcc 使用 `sdcc -v` 能够列出 `alp15`，ALP15 这个选项就会出现在下拉列表里**

---

另外，请提供一下你们的 sdcc 和 示例工程

github0null, 2024-06-04:
本质上这个下拉选项就是为了生成 `-m <processor>` 参数

lihebb, 2024-06-04:
目前SDCC的sdcc -v 已经可以出现/alp14/alp15了。
那我直接等您下一个版本更新即可对吗（预计大概什么时候可以更新）？

非常感谢您的支持。

lihebb, 2024-06-04:
目前我本地的SDCC -V可以出现alp14/alp15了，但是还未Push到open source。您支持之后我可以先用我本地的测试。

github0null, 2024-06-04:
> 目前SDCC的sdcc -v 已经可以出现/alp14/alp15了。 那我直接等您下一个版本更新即可对吗（预计大概什么时候可以更新）？
> 
> 非常感谢您的支持。

你要测试的话不用等我更新，这个配置实际是保存在一个文件里的，你可以直接编辑，把 "device" 改成 alp15，保存就行了

[image]

lihebb, 2024-06-04:
ok，非常感谢。我先按这个方法测试。
等后续测试完成了我再提供SDCC和示例工程给您。

github0null, 2024-06-10:
v3.17.2 预发布版，已经调整了该功能，允许自定义名称

closed 2024-06-10 by github0null

lihebb, 2024-06-15:
ok，非常感谢您的支持