github0null/eide #348: [question]  好像生成的arm-gcc的stm32项目里边的unify_builder无法在Linux上正确定位dotnet runtime

Dawn-Xu-helloworld, 2024-06-08:
dotnet  --list-runtimes
Microsoft.AspNetCore.App 6.0.25 [/opt/dotnet-sdk-bin-6.0/shared/Microsoft.AspNetCore.App]
Microsoft.NETCore.App 6.0.25 [/opt/dotnet-sdk-bin-6.0/shared/Microsoft.NETCore.App]

You must install or update .NET to run this application.

App: /home/dawn/.local/share/Trash/files/cl.eide-3.17.1/res/tools/linux/unify_builder/unify_builder
Architecture: x64
Framework: 'Microsoft.NETCore.App', version '6.0.0' (x64)
.NET location: /home/dawn/.local/share/Trash/files/cl.eide-3.17.1/res/tools/linux/unify_builder/

No frameworks were found.

github0null, 2024-06-08:
你需要正确的安装 dotnet 运行时，unify_builder 本身会检查运行时（这是自动的行为，不是人为编写的逻辑）

如果你的环境变量不正确或者安装位置不正确，可能就会导致该问题

https://learn.microsoft.com/zh-cn/dotnet/core/runtime-discovery/troubleshoot-app-launch?pivots=os-linux

Dawn-Xu-helloworld, 2024-06-09:
> 你需要正确的安装 dotnet 运行时，unify_builder 本身会检查运行时（这是自动的行为，不是人为编写的逻辑）
> 
> 如果你的环境变量不正确或者安装位置不正确，可能就会导致该问题
> 
> **linux 上最好使用包管理器进行安装**
> 
> https://learn.microsoft.com/zh-cn/dotnet/core/runtime-discovery/troubleshoot-app-launch?pivots=os-linux

可是我就是用包管理器安装的dotnet6.0

Dawn-Xu-helloworld, 2024-06-09:
啊,莫名其妙又好了？？？？

closed 2024-06-09 by Dawn-Xu-helloworld