github0null/eide #361: [question] 如何为多个工程指定不同的物理daplink设备?

Antecer, 2024-08-04:
最近在调试一个项目,这个项目有多个外部硬件,于是我就接了多个daplink,然后问题就出现了.
EIDE我找不到如何给每个工程分配不同的daplink, 结果就是只能用一个daplink来回拔插, 是真的累啊.

github0null, 2024-08-05:
你看一下 OpenOCD 有没有相关命令，有的话，需要你自己编写烧录命令

---

我猜测应该可以通过修改你的  daplink.cfg 来选择指定 uid 的烧录器的，每个烧录器都有一个 uid

Antecer, 2024-08-05:
[image]

GPT是这样回答的,我也不知道怎么弄.

[image]

能否在eide工程的配置里面添加这么一行参数呢? 
`"dapSerial": "123456789abcdef"`
然后剩下的工作就由插件自动完成了.该参数为空就按照之前的方式默认调用第一个找到的daplink,如果有值且找不到对应的daplink就输出错误信息.

或者在EIDE的项目菜单里面弄个下拉选项,让用户选择使用哪个daplink也是很好的.

github0null, 2024-08-05:
你把 atlink.cfg 这个文件从 openocd 的安装目录里复制出来，放到项目根目录下（和 *.code-workspace 一个目录下）

然后修改 atlink.cfg 这个文件，在里面填序列号

改完后，在烧录配置里重新选成你修改后的 cfg 文件

----

下面是 openocd 默认的 cmsis-dap.cfg 的内容，序列号是注释了的，你取消注释填序列号就行了

```
# SPDX-License-Identifier: GPL-2.0-or-later

#
# ARM CMSIS-DAP compliant adapter
#
# http://www.keil.com/support/man/docs/dapdebug/
#

adapter driver cmsis-dap

# Optionally specify the serial number of CMSIS-DAP usb device.
# adapter serial 02200201E6661E601B98E3B9

```

github0null, 2024-08-06:
> 测试了,确实可以这么用. 不过我仍然希望能把 `adapter serial` 这个参数直接添加到eide.json的openocd里面,这样就不需要额外管理一个配置文件. 感谢你的工作!

openocd 的烧录流程是靠这个 cfg 文件实现的，

这个cfg文件本质是一个 tcl 脚本文件，很灵活，如果你要修改默认值，增加配置，控制烧录流程，你都需要去自己修改脚本实现，维护自己的脚本是不可避免的

因此这个 GUI 里没有加入诸如：
 - 烧录速度，接口类型，复位类型 等等常用的配置

因为这些配置都可以在 cfg 文件里自己改

Antecer, 2024-08-07:
好吧,这对于大多数只把daplink当个普通下载器来用的人来说,确实略复杂了点.
不过,我已经达到目的了,这就够了.

closed 2024-08-07 by Antecer