github0null/eide #363: [question] 请教一些 eide.json 的配置参数问题

Antecer, 2024-08-16:
这是我的eide项目配置文件:

```json
{
	"name": "ChainSPI_Master",
	"type": "ARM",
	"dependenceList": [],
	"srcDirs": [
		"project",
		"libraries",
		".pack/ArteryTek/AT32F402_405_DFP.2.0.4/Device/Include",
		".pack/ArteryTek/AT32F402_405_DFP.2.0.4/Device/Firmware/Peripherals/inc",
		".pack/ArteryTek/AT32F402_405_DFP.2.0.4/Device/Firmware/Peripherals/src"
	],
	"virtualFolder": {
		"name": "<virtual_root>",
		"files": [],
		"folders": []
	},
	"outDir": "build",
	"deviceName": "AT32F405RCT7-7",
	"packDir": ".pack/ArteryTek/AT32F402_405_DFP.2.0.4",
	"miscInfo": {
		"uid": "c6dd31095745a76c6d7d5df89d1b70c0"
	},
	"targets": {
		"Debug": {
			"excludeList": [
				"libraries/cmsis/cm4/device_support/startup/iar",
				"libraries/cmsis/cm4/device_support/startup/mdk",
				"libraries/cmsis/cm4/device_support/startup/gcc_selected"
			],
			"toolchain": "GCC",
			"compileConfig": {
				"cpuType": "Cortex-M4",
				"floatingPointHardware": "none",
				"scatterFilePath": "libraries/cmsis/cm4/device_support/startup/gcc/linker/AT32F405xC_FLASH.ld",
				"useCustomScatterFile": true,
				"storageLayout": {
					"RAM": [],
					"ROM": []
				},
				"options": "null"
			},
			"uploader": "OpenOCD",
			"uploadConfig": {
				"bin": "",
				"target": "at32f405xx",
				"interface": "${workspaceFolder}/atlink",
				"baseAddr": "0x08000000"
			},
			"uploadConfigMap": {
				"OpenOCD": {
					"bin": "",
					"target": "at32f405xx",
					"interface": "atlink",
					"baseAddr": "0x08000000"
				},
				"Custom": {
					"bin": "",
					"commandLine": "",
					"eraseChipCommand": ""
				}
			},
			"custom_dep": {
				"name": "default",
				"incList": ["."],
				"libList": [],
				"defineList": [
					"USE_STDPERIPH_DRIVER",
					"AT32F405RCT7_7",
					"USB_OTG_HS",
					"__DEBUG"
				]
			}
		}
	},
	"version": "3.4"
}
```

> #### 问题一
> "packDir" 能否作为变量的方式插入 "srcDirs" 的路径中, 这样我就能使用一个eide模板配置文件自适应多种不同的芯片资源,使用的时候只需要切换 "芯片支持包" 就行了, 我想让 "srcDirs" 的配置变成这样:
```
"srcDirs": [
	"project",
	"libraries",
	"${packDir}/Device/Include",
	"${packDir}/Device/Firmware/Peripherals/inc",
	"${packDir}/Device/Firmware/Peripherals/src"
],
"packDir": ".pack/ArteryTek/AT32F402_405_DFP.2.0.4",
```

> #### 问题二
> 构建配置的 "excludeList" 列表,能否纳入到对应构建起的分支配置里面? 这样就能在切换编译器的时候,自动切换排除列表,不在需要手动编辑排除列表.(PS: /startup 的上级目录是所有编译器都需要的公共文件,只有 startup_at32f402_405.s 是每个编译器独立调用,因此我无法单独添加 srcDirs 引用)
> "scatterFilePath" 也是同理,每次切换编译器后都要手动修改 ld 文件的路径,这挺不方便.
> 我设想的解决方案是: 将 "libraries/" 加入 "srcDirs" 列表, 然后在 "excludeList" 排除 ""libraries/cmsis/cm4/device_support/startup/" 文件夹, 然后在 "compileConfig" 节点下添加 "includePath": "libraries/cmsis/cm4/device_support/startup/gcc",  includePath将具有更高的优先级被加入资源引用, 最后还能作为变量使用到配置 "scatterFilePath": "${includePath}/linker/AT32F405xC_FLASH.ld"  从而实现切换编译器后自动切换startup文件路径 和 ld 文件路径:
```json
"excludeList": [
	"libraries/cmsis/cm4/device_support/startup"
],
"toolchain": "GCC",
"compileConfig": {
	"cpuType": "Cortex-M4",
	"floatingPointHardware": "none",
	"startupPath": "libraries/cmsis/cm4/device_support/startup/${toolchain}",
	"scatterFilePath": "${startupPath}/linker/AT32F405xC_FLASH.ld",
	"useCustomScatterFile": true,
	"storageLayout": {
		"RAM": [],
		"ROM": []
	},
	"options": "null"
},
```

github0null, 2024-08-16:
- 问题1
  
  只有 virtualFolder 里可以使用 变量，变量可以在 项目设置->环境变量 里添加；srcDirs 不支持变量

- 问题2

  这个 排除列表 是和 target 绑定的，不是和构建配置绑定

  如果你要使用 两种编译器 ，那么你需要建立两个target，分别适配两种编译器

  编译的时候切换不同的 Target 就行了

github0null, 2024-08-16:
这个分类下有个 nrf52832 的模板，是适配了 两种编译器的，你可以看一下

[image]

Antecer, 2024-08-16:
> * 问题1
>   只有 virtualFolder 里可以使用 变量，变量可以在 项目设置->环境变量 里添加；srcDirs 不支持变量
> * 问题2
>   这个 排除列表 是和 target 绑定的，不是和构建配置绑定
>   如果你要使用 两种编译器 ，那么你需要建立两个target，分别适配两种编译器
>   编译的时候切换不同的 Target 就行了

即便使用 virtualFolder 似乎也没办法自动化的拿到芯片支持包的路径,能否将 "packDir" 作为eide的默认环境变量编译到插件里面呢?

github0null, 2024-08-17:
按照原来的设计：pack 中的内容是通过 GUI 进行安装的，

[image]

没有考虑你的这种用法，即使加了 packDir 这个变量，也是无法实现自动添加所有源文件的，因为 srcDirs 这个属性不支持变量

Antecer, 2024-08-19:
如果能改一下代码,让它支持就方便了

github0null, 2024-08-19:
> 如果能改一下代码,让它支持就方便了

按照现在插件里的做法，很难去更改：带变量的路径只有被使用的时候才会计算出实际的值，但 srcDirs 初始化是要绑定文件监视器的，因此路径一开始就是要固定好，后面是不能变的；有冲突

Antecer, 2024-08-19:
这个变量它本身也是一个静态值来着,我们把它当成"常量"来处理就行了,只要在配置文件加载的时候进行替换,后面应该也不容易出现变更,毕竟一个工程只能有一个支持包, 如果创建新工程需要变更, 那么在变更后重载配置文件应该就行了. 这样就不存在冲突吧?

github0null, 2024-08-19:
这样的单独为这个点 “开小灶” 肯定会引发其他地方的 路径错误 bug，这个插件源码本身就是一堆屎山，有很多历史包袱，而你要修改的部分又是最早期的代码，所以我并不打算去改这个；

> "packDir" 能否作为变量的方式插入 "srcDirs" 的路径中, 这样我就能使用一个 eide.json 模板配置文件自适应多种不同的芯片资源,使用的时候只需要切换 "芯片支持包" 就行了

其他的替代方法：

方法 1. 其实你现在的做法本身就绕过了插件的"芯片支持包"功能，因此不需要用插件"芯片支持包"安装功能；你可以直接把 pack 文件解压，把文件夹重命名成一个 固定的名字 就行了，然后添加到项目，这样也不用每次去修改 srcDir；

方法 2. 使用文件夹软链接，windows 和 linux 都是支持的，要更换 pack 的话，只需重新建立软链接就行了，也不用改 srcDirs 内容

然后把不用的文件用排除功能排除掉，就行了

Antecer, 2024-08-20:
说起这个软连接,我之前编译AD封装库就是用 硬链接 把输出文件链接到库文件夹, 后来用多了过后, 发现每次编译封装库都要在ad里面取消对该库的引用,不然就会产生冲突,后来干脆把编译输出路径加入到AD的封装库引用了,硬链接也不用了...
好吧,看来这个 idea 短时间内是没法达成了,我还是改 srcDirs 吧...