github0null/eide #364: [这个eide的编译相关配置有什么讲究吗]

homohifi, 2024-08-17:
**What you want to do**
A clear and concise description of what you want to do.

**Describe the problem**
A clear and concise description of what the bug is.
我是做飞控的一个姿态解算，用QEKF（就RM的王洪玺）为什么一模一样的工程，姿态解算算法在Keil环境下跑150us，在EIDE下要760us。

**Expected behavior**
A clear and concise description of what you expected to happen.
配置出和Keil下一样快的算法运行效果

homohifi, 2024-08-17:
[image]
这里是我的EIDE配置，可能的可以提供工程以及详细说明

github0null, 2024-08-17:
打开 构建器选项，检查根 优化等级 相关的选项是否跟 keil 上的一样

[image]

如果没有找到问题，你把你导入的 uvprojx 文件发上来

homohifi, 2024-08-18:
> 打开 构建器选项，检查根 优化等级 相关的选项是否跟 keil 上的一样 [...]
[image]
[image]
我不清楚这样对不对

homohifi, 2024-08-18:
然后这个需要发送整个工程吗，还是只要一个uvprojx 文件呀

github0null, 2024-08-18:
只需 uvprojx 文件就行

homohifi, 2024-08-18:
我的uvprojx文件

homohifi, 2024-08-21:
> 只需 uvprojx 文件就行

哥，怎么样了呀

github0null, 2024-08-22:
没看到你的文件，麻烦再发一下

homohifi, 2024-08-22:
[Uploading IMU_with_FREERTOS.zip…]()

homohifi, 2024-08-22:
> 没看到你的文件，麻烦再发一下

收到了吗

homohifi, 2024-08-22:
> 链接是无效的，在论坛上开帖子吧，
> 
> 顺便把你用 mdk 编译后的 map 文件 和 用 eide 编译后的 map 文件也发一下

通过百度网盘分享的文件：map_uvproj
链接：https://pan.baidu.com/s/1ncUY_xCLNbsdHW-r8JbCAg?pwd=2zma 
提取码：2zma 
--来自百度网盘超级会员V5的分享

github0null, 2024-08-22:
两个工程用的库不一样

[image]

eide 上，你需要使用排除功能，把不带 f 后缀的lib排除掉，带 f 的才是你要用的，使用硬件浮点，速度才会变快

[image]

github0null, 2024-08-22:
另外你 keil 里设置的可能有问题，不带 f 的需要排除掉，这样 eide 导入时才会自动设置排除

keil 上编译后没问题，因为你这个两个文件的顺序，带 f 的在前面，缺少的符号在这个库中被找到先被链接了，所以后面那个库直接跳过了

[image]

homohifi, 2024-08-22:
> 两个工程用的库不一样 [...]
解决了，太感谢了！

closed 2024-08-23 by github0null