github0null/eide #369: [question] EIDE AVR SUPPORT ?

Electronic-6502, 2024-08-27:
In some places it is said that EIDE supports AVR, but in some cases it is said that it does not support it, which one is correct?

github0null, 2024-09-09:
First, install avr-gcc compiler on your pc, and register compiler into system PATH

If your system is win32, we can provide a gcc binary:

[image]

Then you can create an example project from internal template:

[image]

setup your mcu name:

[image]

build

[image]

have fun !

----

You can also create a empty project and then copy some example code from: 

https://github.com/hexagon5un/AVR-Programming

[image]

Electronic-6502, 2024-09-09:
Thank you for Answer!

Electronic-6502, 2024-09-11:
Thanks !
I Installed AVR-GCC and Compiled a Project using Em-IDE Plugin , and Nice Worked !
Em-IDE is Very Good Plugin !

closed 2024-09-11 by github0null