github0null/eide #371: [question] 在使用Eide和stm32cubeide配合开发的时候，引入C++标准库头文件会引发vscode报错

FlappyEcho, 2024-09-04:
本人在使用vscode的eide插件配合stm32cubeide进行开发，并希望在项目中引入C++语言的标准库

在通过eide引入cproject以后，若是include C++的标准库头文件就会报错：

基于 configurationProvider 设置提供的信息检测到 #include 错误。
和
无法打开 源 文件 "bits/c++config.h" (dependency of "map")C/C++(1696)

在有些时候会只有第二个错误，并且引发该错误的操作室我引入了#include<map>
且经过尝试，引入其他标准库文件也会报错。

在项目中，我有引入用于C与C++混合开发的相关预处理，如_cppplusplus和extern "C"等。

该报错会直接影响在vscode下相关的代码补全，影响开发体验和效率。
但是即便如此，头文件是可以F12进去的，并且引入了C++标准库的代码也可以在eide和stm32cubeide环境下进行成功的编译和烧录。

eide的说明书有说：不要手动 新建，编写 c_cpp_properties.json 文件
所以在setting.json文件中，我已经注释了所有C_cpp相关的配置
曾有一个解决方案提到如此配置setting.json
"C_Cpp.default.configurationProvider": "ms-vscode.cmake-tools"
这虽然可以解决头文件的问题，但是会引入新的问题，如在C++环境下的hal库调用皆会冒红，所以该解决方案大体不可用。

现在似乎能用的办法就是禁用错误波浪线，并引入configurationProvider的json配置，但是屏蔽错误总归是一件不安全的事，所以我想请问有没有什么办法能解决这个问题？
如果能使用C++的相关语法以及标准库来加入嵌入式开发会大大的提高效率，而且vsc的代码补全更好用，我实在不想放弃这个功能。

FlappyEcho, 2024-09-05:
我在之后在尝试独自解决问题，虽然并没有成功解决。
我在这里补充一些后续尝试的细节。

首先，我的代码本身皆可以通过stm32cubeide内编译器的编译并成功烧录进板子，因此可以基本断定代码本身是没有问题的，不存在什么extern "C"之类的问题。
因为现象略有复杂，所以我起一下标号。

1、完全按着EIDE的手册，不再setting.json内添加任何C_CPP相关的设置。
在这样的情况下，vscode内添加C++的标准库会报错：【无法打开 源 文件 "bits/c++config.h" (dependency of "map")】此处我添加了一个<map>,并且经过尝试添加几乎所有其他的标准库都会报错。
我一路F12进头文件，以图找到该错误具体定位在哪，表现上发现了如下报错路线：
map—>bits/stl_tree.h—>bits/stl_algobase.h
并最终在stl_tree.h中彻底无法找到stl_algobase.h，当然，这两个文件都在同一目录下。
我初步怀疑这是stm32cubeide并没有引入bit的include路径，并且stm32cubeide也确实如此，图下的bits目录是我更新后加的
[image]
**在更新了cubeide的包含路径后，我重新载入eide，确实不存在报错，但是有关标准库的代码自动补全仍然无效。并且在手动的关闭和重新打开VSCode的错误曲线后，报错与上述又重新回归了**
我不知道这是不是因为关闭和开启错误曲线默认增加了setting.json中的c_cpp插件属性，但其实后续会发现，增加有些属性并不会影响eide工程本身。
我尝试在引入了bit的包含路径后，进入map查看为什么没有产生代码补全，发现vscode内__cplusplus的宏设置为201103L，让整个map内的代码失效，但即便是在更改了cubeide内的C++版本，并重新导入到eide中后，eide中的C++版本仍是C++11。如下图所示。
[image]
[image]
后续，我通过很多不是很安全的办法，例如手动设置__cplusplus的宏为201703L，当然，即使map内的文件在vscode查看下成为有效代码了，vscode内仍然无法针对标准库进行代码补全，并且刷新错误曲线后重新报错。

2、在setting.json内适当的修改C_CPP的插件属性
如果在setting.json内引入如下两个设置的任何一个，则可以让vscode `eide识别并补全标准库代码
[image]
但是这似乎就会如eide的说明书所说，增加该插件属性让eide自己的一些配置无法运作，进而导致了在cpp文件内，引入纯C的语法会有报错【而对于使用Hal库的嵌入式开发，纯C是必须的】。
[image]
如上图所示，这是在一个cpp源文件内，使用了cpp特性的代码，这段代码是可以正常编译的，它正确使用了__cplusplus以及extern "C"，可是vscode eide却识别它为报错。【虽然没有什么实际关系，但看着满江红报错终究不舒服】
而似乎唯一的办法就是彻底屏蔽错误曲线，让所有错误都在编译时处理。
但当然，这不安全，也推后了很多错误发现的时间，并不是一个特别完美的解决方案。

不知道我的现象描述是否充分，我希望能够彻底解决这个错误，这样便可以结合vscode强大的代码编辑能力与stm32cubeide方便的嵌入式开发能力，并且引入C++一些更加方便和具有先进性的特性来融入到嵌入式开发中去，令代码更加现代化和易维护。

万分感谢您的耐心，希望可以得到您的指导，又或者在后续的更新版本中荣誉对该现象的处理方案以更方便进行开发环境配置。
再此万分感谢。

github0null, 2024-09-05:
我试了一下，我这边能够正常跳转。

请按照如下步骤，检查 补全配置：

- 确保你的文件是 `c++` 文件（插件通过文件后缀名判断文件类型，并决定使用何种配置）：头文件应该带有 hpp 等后缀，源文件应该带有 .cpp/c++/cxx 等后缀 
- 打开有问题的 c++ 文件
- ctrl+shift+p 打开命令面板
- 执行 `C/C++: Log ...` 日志诊断

[image]

---

1. 一般情况下，不要去修改 C_Cpp.default.configurationProvider，cpptools 通过这个配置提供者获取项目的基本信息
```
"C_Cpp.default.configurationProvider": "cl.eide"
```

  打开输出面板，选择 `eide-cpptools-log` 可以查看 eide 插件向 cpptools 提供的项目信息

2. 系统标头是 gcc 根据编译参数生成的，尽量不要自己添加系统包含目录（自己添加的路径很可能是错误的）

github0null, 2024-09-05:
要更改 __cplusplus 

需要设置c++语言标准，到 构建器配置，c/c++ 编译器选项中进行更改，选完之后保存后，才会生效；

之后检查 eide-cpptools-log

[image]

检查 C/C++ Diagnostics

[image]

FlappyEcho, 2024-09-05:
我一边照做和分析，一边提供一下我这里的相关输出，内容有点多，我消化一下。
1、我这里现象与您大相径庭，首先，<bits/c++config.h>肯定是进不去
当然，在这种状态下，任何标准库的代码补全不能被提供。
2、我现在以C_CPP插件属性仅有如下两个的姿态与您交流
[image]
3、下图是日志诊断
[image]
4、非常抱歉我eide和vscode并没有使用多久，我按着您的说法打开eide-cpptools-log，但我并不知道通过什么ctrl shitf p指令或是别的什么让它来输出
然后我重新通过eide加载了cubeide的elp(简写了)工程，它拥有如下输出
```shell
[init] register CustomConfigurationProvider done !

[folder] provideFolderBrowseConfiguration for 'd:\xxxx\ZY\STM32_PROJECT\407ZET6'
browsePath:
    - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Core/Startup/*
    - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Core/Src/*
    - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Usr/*
    - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Usr/tools/*
    - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Drivers/STM32F4xx_HAL_Driver/Src/*
compilerPath: D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\bin\arm-none-eabi-gcc.exe
compilerArgs:
    - -std=c11
    - -mthumb
    - -mcpu=cortex-m3

[source] cpptools request provideConfigurations for 'd:\xxxx\ZY\STM32_PROJECT\407ZET6\Usr\usr_main.cpp'
[source] provideConfigurations
- from_: 407ZET6:Debug (43fa2cb086dede9f33940d8531ae6949)
  uri:
      $mid: 1
      fsPath: d:\xxxx\ZY\STM32_PROJECT\407ZET6\Usr\usr_main.cpp
      _sep: 1
      path: /d:/xxxx/ZY/STM32_PROJECT/407ZET6/Usr/usr_main.cpp
      scheme: file
  configuration:
      standard: c++11
      includePath:
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Core/Inc
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Drivers/STM32F4xx_HAL_Driver/Inc
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Drivers/STM32F4xx_HAL_Driver/Inc/Legacy
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Drivers/CMSIS/Device/ST/STM32F4xx/Include
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Drivers/CMSIS/Include
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Usr
          - d:/xxxx/ZY/STM32_PROJECT/407ZET6/Usr/tools
      defines:
          - __VSCODE_CPPTOOL
          - DEBUG
          - USE_HAL_DRIVER
          - STM32F407xx
      forcedInclude:
          - C:\Users\Win10\.eide\bin\include\internal_headers\gcc_intr.h
      compilerPath: D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\bin\arm-none-eabi-gcc.exe
      compilerArgs:
          - -std=c++11
          - -mthumb
          - -mcpu=cortex-m3
          - ""

[source] cpptools request provideConfigurations for 'd:\xxxx\ZY\STM32_PROJECT\407ZET6\Usr\usr_main.cpp'
[source] cpptools request provideConfigurations for 'd:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\arm-none-eabi\include\c++\12.3.1\map'
[source] provideConfigurations
- from_: 407ZET6:Debug (43fa2cb086dede9f33940d8531ae6949)
  uri:
      $mid: 1
      fsPath: d:\xxxx\ZY\STM32_PROJECT\407ZET6\Usr\usr_main.cpp
      _sep: 1
      path: /d:/xxxx/ZY/STM32_PROJECT/407ZET6/Usr/usr_main.cpp
      scheme: file
  configuration:
      standard: c++11
[... 89 lines cut ...]
        D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\arm-none-eabi\include\c++\12.3.1
        D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\lib\gcc\arm-none-eabi\12.3.1\include
        D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\lib\gcc\arm-none-eabi\12.3.1\include-fixed
        D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\arm-none-eabi\include
    Defines:
        __VSCODE_CPPTOOL
        DEBUG
        USE_HAL_DRIVER
        STM32F407xx
    Forced Includes:
        C:\USERS\WIN10\.EIDE\BIN\INCLUDE\INTERNAL_HEADERS\GCC_INTR.H
    Standard Version: c++11
    IntelliSense Mode: windows-gcc-arm
    Other Flags:
        --g++
        --gnu_version=120301
Total Memory Usage: 92 MB

------- Workspace parsing diagnostics -------
Number of files discovered (not excluded): 1726
Number of files parsed: 109
···

最后，十分感谢您百忙之中的帮助，EIDE很好用，是现在我感觉最好用的VSCODe嵌入式开发插件，希望能在您的指导下解决该问题，C++提供的编译期特性、RAII、以及标准库数据结构如果能与VSCODE强大的代码补全相结合是一件令人期待的事。
我个人对EIDE使用尚少，印象中我只设置了一个GCC工具链，除此再无设置，并且我的电脑中并没有安装Keil，如果犯了低级的错误我十分抱歉，也希望您予以指导。

github0null, 2024-09-05:
你没有选择正确的 Cpu Type, 您的项目是 F407ZET6，但编译参数显示：cortex-m3

[image]

你需要修改 CPU Type 为 Cortex-M4，这样 gcc 才会生成一条 System Include Path，该目录下 含有 bits/c++config.h

[image]

https://em-ide.com/docs/modules/builder#%E5%9F%BA%E6%9C%AC%E9%85%8D%E7%BD%AE

---

这个 eclipse 导入功能，只能确保源文件能够被导入，并不能兼容 eclipse 的构建配置，因此某些编译选项并不能被导入

你需要编辑 eide 项目配置手动修正不正确的参数

https://em-ide.com/docs/getting-started/import_prj#%E9%85%8D%E7%BD%AE-1

https://discuss.em-ide.com/d/344-new-eclipse

FlappyEcho, 2024-09-05:
我第一时间看到了您的解答，并尝试学习和解决问题，还是有不明白的地方，不好意思。
我先贴出我的warning文件
```
##### Configurations For All Targets #####

//
///// Target: 'Debug' /////
//

Incompatible Args:
    /:
        globalArgs:
            - <MCU> = STM32F407ZETx
            - <CPU> = 0
            - <Core> = 0
            - <Floating-point unit> = com.st.stm32cube.ide.mcu.gnu.managedbuild.option.fpu.value.fpv4-sp-d16
            - <Floating-point ABI> = com.st.stm32cube.ide.mcu.gnu.managedbuild.option.floatabi.value.hard
            - <Board> = genericBoard
            - "<Defaults> = com.st.stm32cube.ide.common.services.build.inputs.revA.1.0.6 || Debug || true || Executable || com.st.stm32cube.ide.mcu.gnu.managedbuild.option.toolchain.value.workspace || STM32F407ZETx || 0 || 0 || arm-none-eabi- || ${gnu_tools_for_stm32_compiler_path} || ../Core/Inc | ../Drivers/STM32F4xx_HAL_Driver/Inc | ../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy | ../Drivers/CMSIS/Device/ST/STM32F4xx/Include | ../Drivers/CMSIS/Include ||  ||  || USE_HAL_DRIVER | STM32F407xx ||  || Drivers | Core/Startup | Core ||  ||  || ${workspace_loc:/${ProjName}/STM32F407ZETX_FLASH.ld} || true || NonSecure ||  || secure_nsclib.o ||  || None ||  ||  || "
            - <Cpu clock frequence> = 16
        cIncDirs: []
        cMacros: []
        cCompilerArgs:
            - <Debug level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.c.compiler.option.debuglevel.value.g3
            - <Optimization level> = undefined
            - <Debug level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.cpp.compiler.option.debuglevel.value.g3
            - <Language standard> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.cpp.compiler.option.languagestandard.value.gnupp17
        sIncDirs: []
        sMacros: []
        assemblerArgs:
            - <Debug level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.assembler.option.debuglevel.value.g3
            - <Include paths (-I)> = undefined
        linkerArgs: []
        linkerLibArgs: []

//
///// Target: 'Release' /////
//

Incompatible Args:
    /:
        globalArgs:
            - <MCU> = STM32F407ZETx
            - <CPU> = 0
            - <Core> = 0
            - <Floating-point unit> = com.st.stm32cube.ide.mcu.gnu.managedbuild.option.fpu.value.fpv4-sp-d16
            - <Floating-point ABI> = com.st.stm32cube.ide.mcu.gnu.managedbuild.option.floatabi.value.hard
            - <Board> = genericBoard
            - "<Defaults> = com.st.stm32cube.ide.common.services.build.inputs.revA.1.0.6 || Release || false || Executable || com.st.stm32cube.ide.mcu.gnu.managedbuild.option.toolchain.value.workspace || STM32F407ZETx || 0 || 0 || arm-none-eabi- || ${gnu_tools_for_stm32_compiler_path} || ../Core/Inc | ../Drivers/STM32F4xx_HAL_Driver/Inc | ../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy | ../Drivers/CMSIS/Device/ST/STM32F4xx/Include | ../Drivers/CMSIS/Include ||  ||  || USE_HAL_DRIVER | STM32F407xx ||  || Drivers | Core/Startup | Core ||  ||  || ${workspace_loc:/${ProjName}/STM32F407ZETX_FLASH.ld} || true || NonSecure ||  || secure_nsclib.o ||  || None ||  ||  || "
            - <Cpu clock frequence> = 16
        cIncDirs: []
        cMacros: []
        cCompilerArgs:
            - <Debug level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.c.compiler.option.debuglevel.value.g0
            - <Optimization level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.c.compiler.option.optimization.level.value.os
            - <Debug level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.cpp.compiler.option.debuglevel.value.g0
            - <Optimization level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.cpp.compiler.option.optimization.level.value.os
            - <Language standard> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.cpp.compiler.option.languagestandard.value.gnupp17
        sIncDirs: []
        sMacros: []
        assemblerArgs:
            - <Debug level> = com.st.stm32cube.ide.mcu.gnu.managedbuild.tool.assembler.option.debuglevel.value.g0
        linkerArgs: []
        linkerLibArgs: []
```
我还是得提前报告一声，就是虽然我之前选用了错误的CPU，但是无论是烧录还是编译，在cubeide下还是在eide下都没有问题。
汇编参数和编译参数应该不会影响编辑器找头文件的事，我就没有管，最后我的选项卡设置如下：
[image]
[image]
[image]
[... 142 lines cut ...]

但是最后问题仍然没有解决：
此时./.vscode下已经没有setting.json
C盘的setting.json内容如下
```json
{
    "editor.fontSize": 20,
    "editor.fontFamily": "Cascadia Mono, 'Courier New', monospace",
    
    "terminal.integrated.defaultProfile.windows": "Ubuntu-20.04 (WSL)",
    
    "cmake.cmakePath": "D:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/bin/cmake.exe",
    "cmake.pinnedCommands": [
        "workbench.action.tasks.configureTaskRunner",
        "workbench.action.tasks.runTask"
    ],
    
    "EIDE.DisplayLanguage": "zh-cn",
    "EIDE.ARM.GCC.InstallDirectory": "D:\\Program Files\\ST\\STM32CubeIDE_1.16.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\\tools",
    "EIDE.STLink.ExePath": "${userRoot}/.eide/tools/st_cube_programer/bin/STM32_Programmer_CLI.exe",
    //"C_Cpp.default.compilerPath": "D:\\Program Files\\ST\\STM32CubeIDE_1.16.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\\tools\\bin\\arm-none-eabi-g++.exe"
}
```

非常不好意思，还需要耽误您的时间

FlappyEcho, 2024-09-06:
昨天我在现有的思路下研究，仍然没有解决问题，但我也发现了您跟我提到的系统包含路径的问题，并且在不同文件中打印C++的log，系统包含路径不一样。
我首先贴上我在cubeide中的系统包含以及项目树。
[image]
该项目中，Core和Driver这两个ide生成目录我并没有在里面添加新文件，所有的新文件都在Usr下。
这是usr_main.cpp中的打印
[image]
只有5个系统包含

这是main.c中的打印
[image]
有3个系统包含

按您的意思，我现阶段的目标应该就是想办法让对应的cpp文件打印出7个系统包含就行了（我不知道您那里为什么是6个）
但目前来说的具体操作我仍然不太知晓

github0null, 2024-09-06:
你可以把 gcc 换到 10.0 再试一下，去官网下载

FlappyEcho, 2024-09-06:
好的，感谢您，我抽空去试试。
只是stm32cubeide本身按说是工具齐全的，理论上我并不需要额外安装别的环境【我的嵌入式开发环境只有stm32cubeide和vscode eide】
我也感觉就按您所说，这是一个不知道为什么eide环境没有完全复制到system include path的问题。并且在手动添加这个路径后可以临时的解决。
如果情况允许，我还是希望能使用轻量的，根治解决方案，即通过eide的接口进行合法设置的范畴下解决该问题。

十分感谢您的帮助，我在您的eide论坛上也发了问题，并且跟踪和截图应该更加详细。
也希望您那里如果对我的问题有什么新的思路也告知我一声。

万分感谢您的耐心帮助。

github0null, 2024-09-06:
System Include 是由你所使用的 gcc 自动生成，跟项目导入无关，eide插件不关心 System Include 有多少，也不会去处理 System Include。因为 gcc 自己本来就知道。

这个 System Include 只是 C/C++插件 分析你的源码所需要的

当 C/C++插件需要获取 System Include 时，这时 eide 会提供两个参数：
 - compilerPath 告知 gcc 的位置
 - compilerArgs 告知 gcc 的基本参数有哪些
```
compilerPath: D:\Program Files\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\bin\arm-none-eabi-gcc.exe
      compilerArgs:
          - -std=gnu++17
          - -mthumb
          - -mcpu=cortex-m4
          - -mfpu=fpv5-sp-d16
          - -mfloat-abi=hard
```

接下来 C/C++插件会通过执行 gcc.exe 导出所有内置的 System Include

```
arm-none-eabi-gcc.exe -std=gnu++17 -mthumb -mcpu=cortex-m4 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -E -v - <nul
```

因为 c 文件 和 cpp 文件的 system include 是不一样的，初步推断可能 C/C++ 插件没有处理好这个（cpp 文件要增加额外的 `-xc++` 参数）

不过这个问题：我已经在最新的 eide 中进行了改进，会根据文件类型提供不同的 compilerPath：是 gcc.exe 还是 g++.exe

你可以试一试 

[snapshot.em-ide.com/3.19.7/](https://snapshot.em-ide.com/3.19.7/)

FlappyEcho, 2024-09-07:
非常感谢您一路上的耐心！
我已经尝试了快照版本，但是很遗憾，日志上的system include并没有改变，无法包含特定头文件的问题依然存在。
但在您的指导下，我发现了问题的症结所在，至少我可以通过在eide引入cubeide项目后手动添加那些system include到eide的include path里来保证编辑的正常进行。
我使用工具的方式是仅在vscode上编辑代码，cubeide上烧录和调试，所以这个临时的手段理论上是安全和可行的。
我也期待着在以后能够彻底解决这个问题，毕竟我是一个强迫症，希望能使用最少的应用工具来完成嵌入式的开发工作。
您依旧给我提供了最简单的解决方案，再次对您表示万分的感谢。
【还说这个issue是该由我关闭还是您来关闭？】

closed 2024-09-07 by FlappyEcho