github0null/eide #376: [bug] Eide extension control by docker

NaveenReddy6, 2024-09-12:
**What are you doing?**
Tell us what you were doing when the bug occurred.
We are using eide extension in VS code and building the hex file for firmware application. To automate we are trying to create the docker image and try to add the repo code into eide project and  execute the build, clean, and flash commands.

**Describe the bug**
A clear and concise description of what the bug is.

We tried multiple ways to build the docker images but we are not able to control the eide extension to automate the build process.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

We need to control eide extension to add and build the application.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu, windows, Windows wine
 - EIDE Version: latest
 - VSCode Version: latest

**Additional context**
some logs, like: `<user_home>/.eide/cl.eide.log`

github0null, 2024-09-13:
Assume your build folder is `./build/Debug`

If you want to auto-build your project, you need a builder.params file, in general it is located at: `./build/Debug/builder.params`.

If you not found this file, you need to generate it first by using eide plug-in:

[image]

And then export your unify_builder: `/xxxxx/.vscode/extensions/cl.eide-3.19.9/res/tools/linux/unify_builder` into SYSTEM PATH, this is the build tool.

Here are some equivalent commands:

- `build`: execute `unify_builder -p ./build/Debug/builder.params` on your terminal to build your project
- `rebuild`: execute `unify_builder -p ./build/Debug/builder.params --rebuild` on your terminal to rebuild your project
- `clean`: execute `rm -rf ./build/Debug/.obj` to delete all intermediate files

If you want to change compiler options, you need to edit builder.params directly, and then rebuild project.

NaveenReddy6, 2024-09-17:
Thanks for the information.

Can builder.params controlled by docker, bcoz we are creating docker image which need to have the environment to execute the eide commands?

github0null, 2024-09-19:
`builder.params` is a json text file, used to pass your project information to the unify_builder.exe,

and you can modify some json fields to control the builder.

al-xjb, 2024-09-26:
is it possible to generate builder params for a project which requires unify_builder to build it,

scenario : I'm trying to install vsocde in  a ubuntu based docker image, and then installed eide extension to that vscode, once i get the code, just like in gui is there any possibility to load the project based on .uvprojx ? in terms of windows, when we load project in vscode the eide.json is having all the file, gcc, .flash id and other resources, is it possible to replicate the same in docker based image when trying to build the code.

as of now loading the code, and inserting hard coded builder params, hard coded eide json. is it possible to generate them dynamically based on .uvprojx file that we're gonna have in the project folder ?

github0null, 2024-09-26:
> is it possible to generate builder params for a project which requires unify_builder to build it,

Sorry, we don't support this.

NaveenReddy6, 2024-11-15:
Hi @github0null 

need a small information regarding the unify_builder, as I am trying to create a pipeline with all the installations to build the application 
tools installed: gcc, dotnet, vscode, eide extensions and setting up the path for unify_builder
but we are trying to excute the command unify_builder -p builder.params
it is giving unify_builder not found, but as per the installation steps it is installed successfully.
Installing extensions...
Installing extension 'cl.eide' v3.19.9...
Extension 'cl.eide' v3.19.9 was successfully installed.

is there any way to install the unify builder and control it without docker in ubuntu os vscode?

NaveenReddy6, 2024-11-18:
Hi @github0null can you please share your inputs.

github0null, 2024-11-18:
> it is giving unify_builder not found, but as per the installation steps it is installed successfully.

You need to export executable files search path by `export PATH=.....`, your unify_builder was installed at a path like this: `/xxx/YOUR VSCODE LOCATION/extensions/cl.eide-3.20.0-universal/res/tools/linux/unify_builder`

---

Um... 

According to my understanding, according to the previous record you are trying to put this plug-in into docker to automate the compilation of your mcu firmware, but this plug-in was not designed with these in mind at the beginning, you may encounter difficulties, because most operations require users to click the UI to complete, there is no command line interface.

Perhaps you should try other alternatives,
this plug-in can generate a Makefile template, it only needs a few modifications to run.
https://em-ide.com/en/docs/advance/export_project#export-gnu-makefile

NaveenReddy6, 2024-11-25:
Hi @github0null ,

Thanks for the alternative suggestion, but right now I am not trying in the docker environment.
we have setup the complete setup along with the required tools installation in a single pipeline along with building the application.
Even export I am able to do in the pipeline.
code used in pipeline:
    variables:
      DOTNET_ROOT: '/dotnet'  # Define variable for dotnet path

    steps:
    - script: |
        sudo apt-get update && sudo apt-get install -y \
          software-properties-common \
          wget \
          curl \
          nano \
          vim \
          make \
          openocd \
          dnsutils \
          unzip \
          build-essential
        sudo rm -rf /var/lib/apt/lists/*
      displayName: Install basic dependencies

    - script: |
        curl -sSL https://packages.microsoft.com/keys/microsoft.asc | gpg --dearmor -o /usr/share/keyrings/microsoft.gpg
      displayName: Add Microsoft GPG key

    - script: |
        sudo add-apt-repository "deb [arch=amd64] https://packages.microsoft.com/repos/vscode stable main"
      displayName: Add Visual Studio Code repository

    - script: |
        sudo apt-get update && sudo apt-get install -y code
      displayName: Install Visual Studio Code

    - script: |
        mkdir -p ./vscode-user-data
        code --no-sandbox --user-data-dir=./vscode-user-data --install-extension ms-vscode.cpptools
        code --no-sandbox --user-data-dir=./vscode-user-data --install-extension cl.eide@3.19.9
        sleep 60
      displayName: Configure VS Code extensions

    - script: |
        sudo ls -lamo /home/.vscode/extensions/cl.eide-3.19.9/res/tools/linux/unify_builder/
        echo $PATH
        export PATH="${PATH}:/home/vsts/.vscode/extensions/cl.eide-3.19.9/res/tools/linux/unify_builder"
        echo $PATH
        sudo chmod +x /home/vsts/.vscode/extensions/cl.eide-3.19.9/res/tools/linux/unify_builder/unify_builder
      displayName: Setup Unify Builder path

    - script: |
        wget https://download.visualstudio.microsoft.com/download/pr/f57cd7db-7781-4ee0-9285-010a6435ef4f/ebc5bb7e43d2a288a8efcc6401ce3f85/dotnet-sdk-6.0.425-linux-x64.tar.gz
        mkdir ./dotnet
        tar zxf dotnet-sdk-6.0.425-linux-x64.tar.gz -C ./dotnet
        env:DOTNET_ROOT=/dotnet
        rm dotnet-sdk-6.0.425-linux-x64.tar.gz
      displayName: Download and install .NET SDK

    - script: |
        wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
        tar -xjf gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2

        rm gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
      displayName: Download and install ARM compiler
 - script: |
        pwd
        ls -lamo
        cd $(Build.SourcesDirectory)/pipelines/
        ls -lamo
        unify_builder -p builder.params
      displayName: 'Make Release Build'

this is the pipeline code which used to install and export the requirements.

Please have a look and suggest.

github0null, 2024-11-26:
I don't have any suggestion.

github0null, 2024-12-23:
Hi !
This issue will be closed now, if there are other issues, will be opened at any time !

closed 2024-12-23 by github0null