github0null/eide #385: [question] pyocd烧写自定义pack时无法选择目标芯片名称

nocodo189753, 2024-10-30:
**What you want to do**
点击烧写按钮能正确下载固件

**Describe the problem**
pyocd 无法在线安装的芯片型号可以通过命令行指定pack文件路径烧写，但是EIDE UI上无法填入具体型号，如果填入到附加命令行中又会被UI配置覆盖导致无法点击烧写。

不支持的型号只能设置成cortex-m，点击下载时命令行如下，烧写失败
[image]

手动执行，去掉UI的设置，烧写成功
[image]

**Expected behavior**
UI上的芯片信号可以手动设置

github0null, 2024-10-30:
你可以在配置文件里添加 pack 路径，然后再点击 UI 就有了

pyOCD 的配置文件格式它的官网上有介绍

[image]

---

如果你需要完全自定义的命令，你需要使用 Shell 模式烧录，可以执行任意的命令，或者 shell 脚本

nocodo189753, 2024-10-31:
按照你的方法可以烧写了。

不过在vscode 工作目录与EIDE工程文件目录不一致时，无法在yaml里使用pack的相对路径，应该是eide解析yaml时使用EIDE工程文件目录，烧写时shell使用的是vscode打开的文件夹目录

github0null, 2024-10-31:
这个插件执行命令时的相对路径是以 **当前工程的根目录（项目文件  \*.code-workspace 所在的目录）** 作为基准

不是配置的存放目录

nocodo189753, 2024-10-31:
是的，yaml文件里以code-workspace文件位置设置pack的相对路径，UI上能正确显示、设置芯片型号，但是点击下载时，pyocd会显示找不到pack包，只能用绝对路径形式。我的工程目录结构如下, vscode打开的是顶层目录
> test_prj
|--src_share_lib
|--prj1
|--|--src
|--|--prj
|--|--|--EIDE
|--|--|--|--eide.code-workspace
|--|--|--|--...
|--|--|--MDK

closed 2024-11-06 by github0null