github0null/eide #386: [question] 不能和 clangd 插件一起工作吗？

Antecer, 2024-11-11:
如题，虽然编译是正确的。
但是各种函数调用和头文件引用都提示找不到目标， 使用ms的C/C++插件就没问题了

closed 2024-11-11 by Antecer

Zyx-A, 2025-02-22:
ms的解决相关讨论：
https://github.com/Microsoft/vscode-cpptools/issues/2499
https://developercommunity.visualstudio.com/t/c-intellisense-request-to-support-sdcc-mcs51-decla/1122536

Zyx-A, 2025-02-24:
> ms的解决相关讨论： [microsoft/vscode-cpptools#2499](https://github.com/microsoft/vscode-cpptools/issues/2499) https://developercommunity.visualstudio.com/t/c-intellisense-request-to-support-sdcc-mcs51-decla/1122536

从这个里面又找到了解决问题的思路：
https://github.com/clangd/clangd/issues/528

最终我的配置方法为，在clangd的项目配置文件中，或用户配置文件中添加如下参数配置：
```
CompileFlags:
    Add: [--target=sdcc]
    Remove: [
            --std-c99,
            -mmcs51,
            --opt-code-speed,
            --iram-size,
            --xram-size,
            --code-size,
        ]

Diagnostics:
    Suppress: []
```

Antecer, 2025-02-24:
我之前是没有创建 `.clangd` 文件导致的，后来创建该文件后，eide会根据target自动把 `CompilationDatabase` 指向目标编译路径，然后所有问题都解决了。因为这里有生成 compile_commands.json 文件，clangd会从该文件里面找到所有配置。

Zyx-A, 2025-02-24:
我是通过如下步骤创建的demo项目：
`新建项目`→`内置项目模板`→`89C52 SDCC Quickstart`

如果不在项目配置文件`.clangd`中，或用户配置文件`$HOME/.clangd/config.yaml`中加入如下这些参数：
```
CompileFlags:
    Add: [--target=sdcc]
    Remove: [
            --std-c99,
            -mmcs51,
            --opt-code-speed,
            --iram-size,
            --xram-size,
            --code-size,
        ]
```

就一定会出现报错提示：
`Use of undeclared identifier 'P00'clang(undeclared_var_use)`、
`Unknown type name '__sfr'clang(unknown_typename)`、
`Expected parameter declaratorclang(missing_param)`、
`Expected function body after function declaratorclang(expected_fn_body)`、
```
Unknown argument: '--std-c99'clang(drv_unknown_argument)
Unknown argument: '-mmcs51'clang(drv_unknown_argument)
Unknown argument: '--opt-code-speed'clang(drv_unknown_argument)
Unknown argument: '--iram-size'clang(drv_unknown_argument)
Unknown argument: '--xram-size'clang(drv_unknown_argument)
Unknown argument: '--code-size'clang(drv_unknown_argument)
Too many errors emitted, stopping nowclang(fatal_too_many_errors)
```
等错误。

ioxera, 2025-05-01:
windows上的vscode clangd插件不加载.clangd文件，日志中没有 “Loaded config from "字符串，在setting.json中手动指定clangd的参数“"--compile-commands-dir=${workspaceFolder}/build/Target 1"可以识别到compile_commands.json文件，且日志中有"Loaded compilation database from"信息，尝试修改.clangd文件中的CompilationDatabase: ./build/Target 1为CompilationDatabase: "./build/Target 1"同样不加载.clangd。

clangd version 19.1.2