github0null/eide #395: [bug] 预处理宏定义截取不全

jjz45, 2024-12-25:
导入keil5工程后，点开**项目属性-预处理宏定义**项，该项截取不全
具体情况为：
对于MICRO_NAME=0这种target 中的 preprocess symbol，等号及后面的数字会被忽略，仅截取了MICRO_NAME，这导致#if MICRO_NAME = NUMBER的语法高亮错误，未测试是否会导致编译结果异常

[image]

手动修改预处理宏定义项为“MICRO_NAME=0” (不包括引号)的形式后，#if MICRO_NAME = NUMBER的语法高亮符合预期，本应该高亮的部分为灰色，只匹配到else情况

[image]

手动修改预处理宏定义项为“MICRO_NAME = 0”(不包括引号)的形式后，所有与宏定义相关的语法高亮都异常，表现为全都高亮，无论是否与这个宏相关，都高亮，无灰色显示的部分

[image]

jjz45, 2024-12-25:
补充：这是我在keil上的宏din定义设置
[image]

github0null, 2024-12-26:
@jjz45 

您需要规范您的写法，不要在 = 号两侧添加多余的空格

---

- 导入 Keil 项目时，宏定义字符串中的 空格 和  逗号 作为分隔符

  比如 `TEST_1=1 TEST_2=2` 和 `TEST_1=1,TEST_2=2` 是等价的

  因此如果你需要在宏定义中使用 空格，你需要使用引号 "" 包含你的定义

  `TEST_1="A B C"` 或者 `"TEST_1=A B C"`

  目前插件按照上述语法进行解析，因此你不能使用 `TEST_1 = 1`，这样做会使定义被拆分为多个部分 `TEST_1`, `=`, `1`

- 另外在插件上将 `TEST_1=1` 修改为  `TEST_1 = 1` 并不会导致编译错误（构建程序会处理这种情况）

  但 C/C++ 插件无法识别 `TEST_1 = 1` 因此语法高亮会出现问题，TEST_1 定义是无效的

jjz45, 2024-12-27:
### 感谢，在keil项目中修改宏定义使其没有多余的空格后能够被正确截取到了

截取到正确的宏定义后，语法高亮也是正常的。

希望能够添加对于多余的空格的支持使其能够导入keil风格的preprocess symbol。这些宏对于keil工程来说往往具备相当重要的意义，尽管因此这些问题很容易在前期发现，但在问题发生时始终是令人困惑的，因为它看似导致了从keil导入的工程与原工程的结果不一致。对于尝试更换开发环境的用户来说，这是一个让人不安的信号。

closed 2024-12-27 by github0null