github0null/eide #396: [bug] 找不到生成的.d文件

aefj11, 2024-12-28:
一、是否可以正确编译
使用mdk可以正常编译
[image]
二、使用eide的相关配置及文件
1.包含的目录
[image]
2.头文件
[image]
3.源文件
[image]
三、编译错误
[image]
四、存在的疑问
因为是初学者，对整体编程不太熟悉，可能犯了比较低级的错误，但是我目前已知的排查只能做到这里，如果问题比较低级，也请麻烦帮帮忙，非常感谢

出现该报错后，首先去排查了源文件是否存在，检查为：存在
排查头文件是否存在，检查为：存在
是否进行了包含操作，检查为：已包含

因为刚开始学，所以怀疑.d文件为将文件编译后生成的文件，尝试过修改目录名称，因目录名称包含中文（尝试修改，未生效）
总之尝试过各种力所能及的办法，没有解决，所以想请您帮帮忙，非常期待及感谢您的回答

github0null, 2024-12-28:
项目名称，target 名称中不要含有中文

aefj11, 2024-12-28:
> 项目名称，target 名称中不要含有中文

确实是中文惹得祸，一直不知道怎么修改，全局搜索修改了下解决了，感谢

closed 2024-12-28 by aefj11