github0null/eide #403: [question] 打开code-workspace文件后，无法搜索到指定的内容

TwoBrushes, 2025-01-13:
1、打开code-workspace后，插件自动识别为mdk project, 但在资源管理器中，看不到源文件，以致于无法搜索到指定的内容
[image]
2、在eide视图中，所有文件都存在
[image]

TwoBrushes, 2025-01-13:
> 参考：https://discuss.em-ide.com/d/319

收到，非常感谢，修改folders会影响.vscode的存放位置吗？会不会创建在项目文件的上级目录中？

github0null, 2025-01-14:
folders 有顺序，你把主目录放到数组第一个就行了；

TwoBrushes, 2025-01-14:
> folders 有顺序，你把主目录放到数组第一个就行了

我按照你的图中的示例修改了，数组第一个为EIDE项目的上一级目录 ，好象.vscode还是和EIDE项目同目录

github0null, 2025-01-14:
> > folders 有顺序，你把主目录放到数组第一个就行了
> 
> 我按照你的图中的示例修改了，数组第一个为EIDE项目的上一级目录 ，好象.vscode还是和EIDE项目同目录

这就不太清楚了，因为这个功能本来属于 vscode 自带功能，可能在某个时间点作了调整。
建议你导入项目的时候就选好位置，这样就不会有这么多的问题

TwoBrushes, 2025-01-14:
> > > folders 有顺序，你把主目录放到数组第一个就行了
> 
> > 
> 
> > 我按照你的图中的示例修改了，数组第一个为EIDE项目的上一级目录 ，好象.vscode还是和EIDE项目同目录
> 
> 
> 
> 这就不太清楚了，因为这个功能本来属于 vscode 自带功能，可能在某个时间点作了调整。
> 
> 
> 
> 不确定你需要修改 .vscode 中的什么。如果是调试配置，你可以按照这个文档操作：
> 
> https://code.visualstudio.com/docs/editor/multi-root-workspaces#_debugging
> 
> 
> 
> ---
> 
> 
> 
> 建议你导入项目的时候就选好位置，这样就不会有这么多的问题

好的，谢谢

closed 2025-01-15 by github0null