github0null/eide #404: [question] 有没有办法一次性把项目的多个目标依次编译出来？

Antecer, 2025-01-14:
如题，因为某个项目涉及到多个不同的产品，于是每次都要 切换目标&构建 重复操作很多次，感觉太繁琐了。
有没有一种办法，让工程里的所有目标依次完成全部构建呢？

github0null, 2025-01-14:
这样的话只能通过编写脚本来依次执行 `unify_builder -p ./build/xxx/builder.params --rebuild` 来执行编译

每个 target 的 builder.params 需要预先生成好，都存放在 build/<target名> 的目录下

---

builder.params 只有在添加了新的源文件，或者修改了编译参数后，才需要重新生成一次

Antecer, 2025-01-14:
这样预先生成 builder.params 的方式可能因为clean给删掉吧？然后又是繁复的重复劳动了。

eide插件，能否通过 vscode task 发送命令的方式切换 target&rebuild 呢？
target 列表的话，可以通过遍历 eide.json 获得。

比如这样的流程：
1. 遍历 eide.json 获得 target 列表
2. task 发送vscode命令让eide切换target
3. task 发送vscode命令让eide构建target
4. 重复2和3直到所有的target构建完成。

github0null, 2025-01-14:
这些vscode命令的方式暂时实现不了，因为很多操作是异步的，因此 task 是无法顺序执行的；

目前的版本只能通过上面的命令来执行，builder.params 可以移动到任意位置，也可以重命名，可以放在别的地方

Antecer, 2025-01-14:
好吧， 希望EIDE以后能实现一下一次性编译多个目标的功能。

owen520, 2025-04-14:
折腾了半天，原来是不支持一键编译所有目标，这个功能很多人有这个需要的，我有一个keil工程，20多个target，一个一个去点就太烦人了

> 这样的话只能通过编写脚本来依次执行 `unify_builder -p ./build/xxx/builder.params --rebuild` 来执行编译
> 
> 每个 target 的 builder.params 需要预先生成好，都存放在 build/<target名> 的目录下
> 
> 在写好脚本后，比如：`build_all.sh`，可以新建一个 vscode task, task 类型名为：`eide.bash`
> 
> 然后用这个 task 去执行一键编译所有
> 
> 添加了新的源文件，或者修改了编译参数后，builder.params 需要重新生成一次

Antecer, 2025-04-18:
> 折腾了半天，原来是不支持一键编译所有目标，这个功能很多人有这个需要的，我有一个keil工程，20多个target，一个一个去点就太烦人了

我已经使用 xmake 作为构建器了...

closed 2025-04-18 by Antecer