github0null/eide #409: [bug] Ctrl+Click 头文件跟踪路径异常

Antecer, 2025-02-10:
[image]

如图，当我尝试在*.c文件里面跟踪一个头文件，它的路径丢掉了根目录，从而导致vscode定位不到这个文件。
但是，奇怪的是，我在同级目录的*.h文件里面跟踪一个头文件，它又能正确的跳转到对应的头文件(没有丢掉根目录)。

github0null, 2025-02-10:
这个插件不提供代码浏览相关的辅助功能

跟你使用的 其他插件相关，比如 Clangd 和 C/C++ 插件

- 如果你使用 Clangd 请检查 `.clangd` 配置文件是否正确配置，以及检查 build 目录下的 compile_database.json 中的 `at32f402_405_spi.c` 对应的条目是否包含这个 .h 所在的目录
 
- 如果你使用的是 C/C++ 插件，请打开输出面板 检查 `eide-cpptools-log` 中的 IncludePath 是否包含这个 .h 所在的目录

Antecer, 2025-02-10:
我用的 .clangd 插件，我检查下是什么情况吧。
你说的 compile_database.json  是指  compile_commands.json 吗？我这里只有这个文件

Antecer, 2025-02-10:
搞定了，给 .clangd 文件添加以下配置就行了。
```
CompileFlags:
    Add:
        - -I.
```

closed 2025-02-10 by Antecer