github0null/eide #424: 编译成功，烧录一直报错，但是keil5烧录没问题

yaofangwx123, 2025-05-11:
之前是可以的，最近我修改了一些东西不知道哪里出问题了，请指教一下，非常感谢

Active code page: 437
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.8.0
      -------------------------------------------------------------------

Error: No debug probe detected.

 *  终端进程“cmd.exe /C "chcp 437 && C:\Users\10493\.eide\tools\st_cube_programer\bin\STM32_Programmer_CLI.exe -c port=SWD freq=4000 --download "d:\soft\hard\keil5MDK\project\projects\Test\build\Target 1\MiyoAutoTest.hex" -v --go"”已终止，退出代码: 1。 
 *  终端将被任务重用，按任意键关闭。 

{
  "name": "MiyoAutoTest",
  "type": "ARM",
  "dependenceList": [],
  "srcDirs": [],
  "virtualFolder": {
    "name": "<virtual_root>",
    "files": [],
    "folders": [
      {
        "name": "User",
        "files": [
          {
            "path": "User/main.c"
          },
          {
            "path": "User/stm32f10x_it.c"
          }
        ],
        "folders": []
      },
      {
        "name": "Startup",
        "files": [
          {
            "path": "Libraries/CMSIS/startup_stm32f10x_md.s"
          }
        ],
        "folders": []
      },
      {
        "name": "Stdperiph_Driver",
        "files": [
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_gpio.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_rcc.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/misc.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_tim.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_usart.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_iwdg.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_adc.c"
          },
          {
            "path": "Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_i2c.c"
          },
          {
[... 328 lines cut ...]
            "use-microLIB": true,
            "output-debug-info": "enable"
          },
          "c/cpp-compiler": {
            "optimization": "level-0",
            "one-elf-section-per-function": true,
            "c99-mode": true,
            "C_FLAGS": "--diag_suppress=1 --diag_suppress=1295",
            "CXX_FLAGS": "--diag_suppress=1 --diag_suppress=1295",
            "warnings": "all-warnings",
            "gnu-extensions": true
          },
          "asm-compiler": {},
          "linker": {
            "output-format": "elf",
            "xo-base": "",
            "ro-base": "0x08000000",
            "rw-base": "0x20000000"
          }
        }
      }
    }
  },
  "version": "3.5"
}

yaofangwx123, 2025-05-11:
错误: 任务“eide.flasher: STLink”既不指定命令，也不指定 dependsOn 属性。将忽略该任务。其定义是:
{
    "type": "shell",
    "id": "shell,\"chcp 437 && C:\\Users\\10493\\.eide\\tools\\st_cube_programer\\bin\\STM32_Programmer_CLI.exe -c port=SWD freq=4000 --download \"d:\\soft\\hard\\keil5MDK\\project\\projects\\Test\\build\\Target 1\\MiyoAutoTest.hex\" -v --go\",",
    "problemMatcher": [],
    "label": "eide.flasher: STLink"
}

github0null, 2025-05-13:
> Error: No debug probe detected.

就是没有连接到烧录器，这个你得自行排查，与你的项目文件无关

yaofangwx123, 2025-05-13:
搞错了，我之前一直用的是keil assist，我说怎么之前一直可以。

closed 2025-05-13 by yaofangwx123