github0null/eide #429: [bug] 'unify_builder' is not recognized as an internal or external command, operable program or batch file.

Atul-8, 2025-06-16:
**What are you doing?**
我在安装esp-idf 插件后, eide莫名奇妙就用不了了. 
编译会提示这个报错'unify_builder' is not recognized as an internal or external command,
operable program or batch file.
我尝试了卸载重装eide插件和更换版本都无法解决这个问题.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10 xxx]
 - EIDE Version [e.g. v3.8.3]
 - VSCode Version [e.g. v1.68.1]

**Additional context**
some logs, like: `<user_home>/.eide/cl.eide.log`

github0null, 2025-06-16:
找不到命令是因为环境变量被破坏了，也许是别的插件覆盖了进程内的环境变量，因为所有的插件运行在同一个扩展宿主内，也就是同一个进程内。这个你需要禁用相关的插件。重启 vscode 就能解决。

Atul-8, 2025-06-16:
> 找不到命令是因为环境变量被破坏了，也许是别的插件覆盖了进程内的环境变量，因为所有的插件运行在同一个扩展宿主内，也就是同一个进程内。这个你需要禁用相关的插件。重启 vscode 就能解决。
已经找到解决方法了  : 重装一下window的驱动库还是啥的东西就可以了
 https://learn.microsoft.com/zh-cn/cpp/windows/latest-supported-vc-redist?view=msvc-170

closed 2025-06-16 by Atul-8