github0null/eide #432: [bug] STM8 COSMIC Build Failure in EIDE 3.21.4

kiavash599, 2025-06-29:
Regression: STM8 COSMIC .o Files Not Generated in EIDE v3.21.4
Description of the Issue
After upgrading Embedded IDE (EIDE) to version 3.21.4, building any STM8 project using the COSMIC toolchain no longer produces the expected .o object files. The build process runs through all C source files (the compiler doesn’t show any obvious errors during compilation), but no .o files appear in the build/STM8S003/.obj directory. Consequently, the linking stage fails with errors about missing object files. For example, the linker throws an error like:
bash
Copy
Edit
#error clnk .\build\STM8S003\stm8.lkf:52 can't open file .\build\STM8S003\.obj\libraries\STM8S_StdPeriph_Driver\source\stm8s_adc1.o
This regression did not occur in EIDE v3.21.3. Using the exact same project configuration and toolchain, EIDE 3.21.3 generates all .o files properly and the project links successfully. Nothing in the project setup (source code, project settings, COSMIC compiler installation, etc.) was changed apart from the EIDE extension update.
Steps to Reproduce
Open or create an STM8 project in VSCode with EIDE v3.21.4 using the COSMIC STM8 toolchain.
Build the project (using EIDE’s build command).
Observe the build output: the compiler appears to compile all source files without fatal errors, but the *.o object files are not present in the build/.../.obj folder as they should be.
The linker then fails, reporting that it “can’t open” each expected .o file (as shown in the error above).
Now, downgrade the EIDE extension to v3.21.3 and rebuild the same project (same eide.json configuration).
In v3.21.3, the .o files are generated correctly in build/STM8S003/.obj, and the linking completes without errors. This confirms the issue is specific to changes in v3.21.4.
Expected Behavior
In EIDE v3.21.4, building an STM8 project with the COSMIC compiler should produce .o object files for each source file in the designated output folder (e.g. build/STM8S003/.obj). The linker should then find all object files and produce the final binary without errors – just as it works in v3.21.3.
Actual Behavior in v3.21.4
No .o files are being generated in the build/.../.obj directory during compilation.
The build log shows each source file being “compiled”, but there is no corresponding object file output.
The linker fails because it cannot find the object files (resulting in errors like the example above).
There are no other error messages during compilation to indicate why the object files aren’t produced – it simply seems like the output path is being ignored or skipped for COSMIC in this version.
Additional Information
Environment: Windows 10 (64-bit) with COSMIC STM8 compiler (Free Special Edition). The issue is within VSCode using EIDE extension v3.21.4.
Project Configuration: The eide.json project configuration is unchanged between versions. All include paths, builder settings, and toolchain configurations are identical. (The same eide.json works perfectly in v3.21.3, but not in v3.21.4.)
Attempted Workaround: I attempted to manually specify the output directory for object files by adding "-co build/STM8S003/.obj" to the COSMIC compiler extra-flags in the EIDE project settings. However, after saving this change, EIDE removes that flag upon reloading the project (perhaps the extension sanitizes or overrides certain flags). In any case, this manual flag addition did not persist or solve the issue.
Current Status: The only reliable workaround so far is to roll back to EIDE v3.21.3, where the STM8 COSMIC build works as expected. This is how I’ve confirmed that v3.21.4 introduced the regression. I have not changed the COSMIC compiler version or any code – the difference in behavior is solely between EIDE extension versions.
Request for Investigation
Could the maintainers please investigate what changed in the build process or file output handling for the COSMIC STM8 toolchain in v3.21.4? It appears that something in the update may have altered how object file output paths are managed or how the COSMIC compiler is invoked, resulting in object files not being written out. Specifically, please verify:
Any changes in build logic or default flags for the COSMIC STM8 toolchain between 3.21.3 and 3.21.4.
Any modifications to how the .obj output folder is handled or how the -co (or equivalent) flag is passed to the COSMIC compiler in the new version.
This regression is reproducible consistently and is blocking STM8 development when using the latest extension version. I’m happy to provide additional logs or test any patches – just let me know what could help. Thank you for looking into this issue! 🙏 Thank you for your great work on this extension. Aside from this issue, it’s been extremely useful for our projects. I’m reporting this in hopes of helping improve EIDE. Please let me know if any more information is needed to pinpoint the cause of this regression.

github0null, 2025-07-01:
I published a new version `v3.21.5` to fix it.

closed 2025-07-01 by github0null

kiavash599, 2025-07-02:
It works well, thank you. :)