github0null/eide #435: [question] Unable to change CompilationDatabase in .clangd file

LovelyA72, 2025-07-10:
**What you want to do**
I wish to change the CompilationDatabase generation path for clangd because PY32's template calls for a different structure and clangd is unable to locate the json file

**Describe the problem**
Every time I edit the CompilationDatabase path, it got reverted back to `./build/Project`. And there's no other way to change it.

**Expected behavior**
Honour the user's choice of CompilationDatabase path instead of forcing it to a hard-coded value.

github0null, 2025-07-10:
有两个方法可以解决这个问题

- 在工程目录下修改 `.clangd` 配置文件中的 CompilationDatabase 的搜索路径使其指向 `build/xxxx`，https://clangd.llvm.org/config#compilationdatabase
- 你可以使用构建后任务来复制这个文件到你想要的路径

LovelyA72, 2025-07-11:
I got it to work by creating a pre-compile task.

[image]

However, generated json file is still kinda buggy and confused clangd a lot.

For example it complained that it was not able to find stdio.h

[image]

The issue resolved when I added
```
CompileFlags:
    Add:
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\arm-none-eabi\include\newlib-nano
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\arm-none-eabi\include\c++\10.2.1
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\arm-none-eabi\include\c++\10.2.1\arm-none-eabi\thumb\v6-m\nofp
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\arm-none-eabi\include\c++\10.2.1\backward
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\lib\gcc\arm-none-eabi\10.2.1\include
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\lib\gcc\arm-none-eabi\10.2.1\include-fixed
        - -Ic:\users\kashouryo\.eide\tools\gcc_arm\arm-none-eabi\include
```
to my `%LocalAppData%\clangd\config.yaml` file.

This is definitively an edge case but since the EIDE project directory is not the parent folder of those source files, clangd is not reading the `.clangd` file in the project directory. Puya's FAE should find a better way to structure their demo projects.

closed 2025-07-11 by LovelyA72