github0null/eide #456: [bug]

yyShine760, 2025-08-08:
描述：使用eide编译stm32工程报错上一次报错unify_builder不是可识别的符号……，这次报错如下。确认未修改过环境变量，插件也仅有C/C++插件与EIDE插件之外再无其他插件。

*  正在文件夹 XXX 中执行任务: "unify_builder -p c:\Users\XXX\Desktop\MagnetoopticalSwitch\MagnetoopticalSwitch\build\MagnetoopticalSwitch\builder.params" 

文件名、目录名或卷标语法不正确。

 *  终端进程“cmd.exe /C ""unify_builder -p c:\Users\XXX\Desktop\MagnetoopticalSwitch\MagnetoopticalSwitch\build\MagnetoopticalSwitch\builder.params""”已终止，退出代码: 1。 
 *  终端将被任务重用，按任意键关闭。

环境：
EIDE 3.23.13
Windows 11 Iot LTSC
vscode
版本: 1.103.0 (user setup)
提交: e3550cfac4b63ca4eafca7b601f0d2885817fd1f
日期: 2025-08-06T21:40:10.271Z
Electron: 37.2.3
ElectronBuildId: 12035395
Chromium: 138.0.7204.100
Node.js: 22.17.0
V8: 13.8.500258-electron.0
OS: Windows_NT x64 10.0.26100

kepler-444b, 2025-08-09:
我换到旧版本的vscode，这个问题解决了

kepler-444b, 2025-08-09:
EIDE昨晚上更新了一个新的版本，有没有朋友试一下，有没有修复呢

github0null, 2025-08-11:
插件已经更新到了 v3.23.14 用于暂时修复该问题

closed 2025-08-11 by github0null